libplctag/libplctag.NET #350: Booleans in UDTs

airwedge1, 2023-09-22:
When you get tag info from @udt tag name with the UdtInfoPlcMapper and you have booleans defined in the UDT you get some weird behavior.  You get a Sint tag that sometimes starts with ZZZZZZZ appended with the name of the UDT, but other times I have seen it start with __bitHost.  CLX will only allocate memory in 4 byte blocks it seems, but if there are multiple booleans it will try to slam them all in together and create a fake tag to continue maintain the 4 byte block off sets.  

I ended up filtering these out by grouping the tags by the offset and then removing that tag where BitNumber==NULL.

What a huge pain, but seems to do what I want now.  

I only opened this up so if anyone else has the same issue it will hopefully save them some time.

kyle-github, 2023-09-23:
This isn't quite correct.   Rockwell does pack boolean fields.   Note that boolean arrays are packed differently.  Single booleans are packed into 8-bit fields (effectively a SINT).  These fields are named with the leading ZZZZ that you noticed.   They are not aligned to 4 byte boundaries.  They are aligned as single bytes.  There may be padding following the ZZZZ field if the following field is aligned at something larger than a single byte.   A boolean field followed by an INT field will take up 4 bytes total and have a single leading byte (SINT) where the boolean is allocated, a padding byte, and then two bytes for the INT field.

Note that the packed byte used for the booleans will be immediately preceding the first boolean in it.

Boolean arrays are another whole ball of crazy.  If they are stand alone tags they are rounded up to the nearest multiple of 32 in size.   In other words they are allocated in chunks of DINTs.   However, when you try to read them, you need to pass the _DINT_ count as the number of elements to read.

kyle-github, 2023-09-23:
Rockwell publication 1756-PM020G-EN-P describes the packing and protocol in some detail.

timyhac, 2023-10-29:
Thanks @airwedge1 and @kyle-github for the information!

closed 2023-10-29 by timyhac