libplctag/libplctag.NET #355: UDT with mixed read/write access

airwedge1, 2023-11-16:
A new interesting issue arose.  I have a UDT where some of the tags within it are tag access = Read/Write and some are tag access = None.  The @udt special tag returns all of the fields regardless of access level.

I'm trying to determine in the @tags or @udt special tags which of the fields have access and which do not.  I looked at and tried debugging UdtInfoPlcMapper but I could not find anything in there that seems to change when the access level is changed.  

Any help would be appreciated.

kyle-github, 2023-11-16:
Huh...   Yeah, that is a problem.   I _think_ that there is an attribute that will tell us the protection on a field, but I am not sure.   Let me reach out to the gurus who might know.

airwedge1, 2023-12-08:
> CAUTION: This email originated from outside of the organization. Do not click links or open attachments unless you recognize the sender and know the content is safe.
>
> My guru contact is not responding. I will keep seeing if there is something I can query to find this out.

Thanks for your time.  Much appreciated.

[created with MySignature.io]<https://kubicacorp.com/>
Nicholas Walker
Software Leader | Kubica
mobile:  248.924.1507
website: Kubicacorp.com<https://kubicacorp.com/>
email:    ***@***.******@***.***>