libplctag/libplctag.NET #356: Expose more TagOfT properties to ITag interface

Devqon, 2023-11-24:
We use this driver with a generic wrapper to expose the tag functionality. We therefore have a method that looks like this:

```csharp
public static ITag MapToLibPlcTag(Type type)
{
    return type switch
    {
        Type _ when type == typeof(bool) => new TagBool(),
        Type _ when type == typeof(int) => new TagDint(),
        // .. etc.
        _ => throw new NotSupportedException();
    };
}
```

This exposes the `ITag` to our wrapper, but we do need some of the functionality that is 'hidden' in the `TagOfT<>` for polling mechanisms, like the `AutoSyncReadInterval` and all of the events (`ReadStarted`, `ReadCompleted` etc.).

There is no way to expose a `TagOfT<>` instead of the `ITag`, because that needs a compile-time generic where we only have a runtime `System.Type`.

Is there a reason that these properties aren't exposed through the `ITag` interface?

timyhac, 2024-06-11:
Sorry for not replying to this - I think your detective work is correct also, I will add the extra events and properties shortly.

timyhac, 2024-06-15:
Hi @Devqon - this is completed in libplctag.NET v1.4.0

closed 2024-06-15 by timyhac