libplctag/libplctag.NET #357: String tag read but not write

Daniel-Dorneles, 2023-12-06:
I have a somewhat strange situation related to tag's string.

I have 3 different PLC projects, in one of them the string variable is within a program scope within the PLC, and in the other two projects, the string variable is within a UDT scope within the PLC, if I can call it that.

It turns out that in the first project the string variable works normally both when reading and writing, in the other two projects only reading works, I even receive value changes coming from the PLC in the ValueChanged event.

Could it be something related to where the string variable is?

I tested several tag creation configurations without success, the strange thing is that the same code works in the first project.

Note: String variables are read and write.

Project 1 (Read OK | Write OK):

[image]

                if (myTag1 == null)
                {
                    myTag1 = new NotifyValueChangedTag<StringPlcMapper, string>()
                    {
                        Name = "PROGRAM:OP10.BARCODE",
                        Gateway = "10.1.44.174",
                        Path = "1,0",
                        PlcType = PlcType.ControlLogix,
                        Protocol = Protocol.ab_eip,
                        AutoSyncReadInterval = TimeSpan.FromMilliseconds(500),

                    };
                    ((NotifyValueChangedTag<StringPlcMapper, string>)myTag1).ValueChanged += ValueChanged;
                    myTag1.Initialize();
                }
                else
                {
                    myTag1.Value = DateTime.Now.ToString();
                    myTag1.Write();
                }

Project 2 (Read OK | **Write NOK**):

[image]

                if (myTag1 == null)
                {
                    myTag1 = new NotifyValueChangedTag<StringPlcMapper, string>()
                    {
                        Name = "SCADA2.BARCODE",
                        Gateway = "10.1.44.43",
                        Path = "1,0",
                        PlcType = PlcType.ControlLogix,
                        Protocol = Protocol.ab_eip,
                        AutoSyncReadInterval = TimeSpan.FromMilliseconds(500),

                    };
                    ((NotifyValueChangedTag<StringPlcMapper, string>)myTag1).ValueChanged += ValueChanged;
                    myTag1.Initialize();
                }
                else
                {
                    myTag1.Value = DateTime.Now.ToString();
                    myTag1.Write();
                }

When clicking to write the error occurs and ErrorBadConfig returns

[image]

I tried several other ways to create the tag, including using string[] and defining ArrayDimensions, but without success.

I appreciate any help.

kyle-github, 2023-12-06:
What is the definition for the UDT string?   I see in the screen shot that there is "/20" in the name.  Not sure what that means.  It is relatively common to define new string types that are shorter than the default type.  They look like strings and the act like strings, mostly.

kyle-github, 2023-12-06:
There is a tool called list_tag_logix that comes with the core C library.   If you cannot get the definition some other way, you could try that tool.  The output is not the greatest, but it should be enough to determine the type.  

Also, is it possible that read/write privs on that UDT field only allow reads?

Daniel-Dorneles, 2023-12-06:
> What is the definition for the UDT string? I see in the screen shot that there is "/20" in the name. Not sure what that means. It is relatively common to define new string types that are shorter than the default type. They look like strings and the act like strings, mostly.

[image]

The tag is defined as read/write, so much so that I can write to it using an opc client. The tag address is SCADA2.BARCODE.DATA/20, but I don't think that's the name you should use. To give an example, in project 1, where reading and writing worked using LibPlcTag, the tag has the address PROGRAM:OP10.BARCODE.DATA/24, but I just used PROGRAM:OP10.BARCODE, and it worked normally.

[image]

35102 refers to which type?

timyhac, 2024-06-11:
35102 refers to the Barcode's type which is going to be some sort of string that is defined in the PLC. Barcode is a field within an instance of some other UDT.

I think the issue you have about access is going to be at the PLC level. Access can be declared within the PLC controller to withhold access to different variables.

timyhac, 2024-07-17:
@Daniel-Dorneles I trust that you were able to work through this question and find a way to write to that string. If you do have more questions please raise a new ticket.

closed 2024-07-17 by timyhac