libplctag/libplctag.NET #358: Issue's when controller is located in last slot of rack

oOICEMANOo, 2023-12-11:
The setup I'm working with is not ideal and has the ENBT in slot 15 and a 1756-L62 in slot 16, I've tried a few different path settings thinking 1,16 should work but gives a bad gateway error. is it possible to connect using the higher slots.

[image]
[image]

kyle-github, 2023-12-11:
Oops, just realize that this in in the .NET project.   I need to move it to the core library.

kyle-github, 2023-12-11:
Moved the comments over to issue [#443](https://github.com/libplctag/libplctag/issues/443) in the core project.

kyle-github, 2023-12-11:
Will update here when I have a version of the core project in which this is fixed.

oOICEMANOo, 2023-12-12:
Thanks for you help! I appreciate your time getting this worked out. Let me know if there's anything I can assist with.

timyhac, 2024-06-15:
Hi @oOICEMANOo - some changes have been made to the core library which will probably have resolved this issue. This is available in the versions of libplctag.NET that have been published today.

kyle-github, 2024-06-15:
This isn't fixed yet.  It depends on [462](https://github.com/libplctag/libplctag/issues/462).