libplctag/libplctag.NET #360: Reading PLC 5 Data File

oOICEMANOo, 2023-12-18:
When trying to read a PLC5/40B using DHRIO I can get values using say B17:5/1 but when I try to read if a output is on ex. O:051/3 I get a ErrorBadParam. Is it possible to read the output file?

CIP Path: 1,2,A:5:5

[image]

[image]

timyhac, 2024-06-15:
Hi @oOICEMANOo - I have never used PLC5 but you may get more luck by adding detail to your question such as suggested in the [Contributions guidance](https://github.com/libplctag/libplctag.NET/blob/master/CONTRIBUTIONS.md#guidance-when-creating-an-issue).

kyle-github, 2024-06-15:
This is not fixed yet but there is an [issue](https://github.com/libplctag/libplctag/issues/465) tracking it.

Also note that all numbers must be _DECIMAL_ only.  Octal is not currently supported.

raentzmi, 2024-06-15:
Couple notes when reading this thread.  PLC 5's were some time ago...

The format of the Input and Output if you did not know is Rack-Module-  so
the example of O:051/3 is Rack 05 Module1 bit 3 - of which all three
numbers are in Octal.

For Logix the use of a period for the bit address has always been used -
for PLC 5 and SLC the \ has been used.  Not sure how the library does this
or not.

I find it interesting that CIP routing is available and that's Great!

In the short term I would suggest moving the Output table to a file of your
choice (i.e. BXX) then use this file instead of the I/O tables.  This opens
up some interesting things you can do with this new table - look at FBC and
DDT.

The PLC 4 is synchronous with the I/O scan so I would do this at the top of
the program for the input table so you have the current state of the inputs
and do the outputs at the end of the program.

If you do this you only need to worry about what outputs changed state in
the last scan - and the I/O that is part of any block transfers if you have
them.

In the case of block transfers you should copy the results out on the done
bit - this is a good practice as the DN bit signals that the contents of
the block is valid.  If you don't do this you can execute logic on data
that is invalid.

Rob Entzminger

kyle-github, 2024-06-15:
Hmm, @raentzmi, that is the most cogent argument I've heard for a while about why to support octal.  Is there something that lets you know the number is octal?  Leading zero?  If so, I will prioritize the parsing of data-file style names.

I did only a little programming of PLC/5 systems.  Almost all the times I touched them it was converting a PLC/5 to a ControlLogix.   Or a PLC/5-250 to a ControlLogix (every tried to find parts for those things?).

You can get to the O file, but you need to put in the data file number, which is zero.  Similarly the I input data file needs the data file number which is 1 (one). 

I actually have a PLC/5 and a MicroLogix, so I can do some testing eventually.

raentzmi, 2024-06-16:
It's my understanding that Input and Output data types can only be Octal.
I want to also say that you can't have an O20:051/10 - meaning Output data
type file 20...

The use of Octal would be a good thing to have but I would say it's not
required.  Is this a bug? I would say no since there is a workaround to
move the Output file data to a Binary file that can be accessed in the
current form of the library.

Depending on how much of the data table is needed the rungs to do this are
very simple - I would say simpler than adding octal to the library.

Knowing that there is a limited install base of PLC 5 (this includes
PLC-5250) and SLC this would see limited useability.

I will see if I can find a PLC 5 and Logix 5 to test this and maybe write a
document on what I find...

Rob Entzminger

timyhac, 2026-11-15:
Hi @kyle-github - its unclear to me what this issue is tracking now. https://github.com/libplctag/libplctag/issues/465 was closed but there is talk of additional features to add to the core library.

kyle-github, 2026-11-15:
The mnemonic field names like .ACC, .DN etc. are not correctly mapped.  I am fixing that.