libplctag/libplctag.NET #363: Doubt abou getting values from a unknown tag

contibru, 2024-02-05:
Hello, I am starting to work if this library and I have one doubt about getting values from a variable which I don`t know the type.

As I see on the library, whenever I want to get a value I should know what is the Type of the variable, but when I don`t it?

Is there a way to get the value as a String no matter the type is?

I give a try on the example called _ExampleListTags_  but I did not see my variable in there, and when I try to search for exatcly the name of the variable, I got this error bellow:

```
libplctag.LibPlcTagException
  HResult=0x80131500
  Message=ErrorOutOfBounds
  Source=libplctag
  StackTrace:
   at libplctag.NativeTagWrapper.ThrowIfStatusNotOk(Nullable`1 status) in E:\Harro\libplctag.NET\src\libplctag\NativeTagWrapper.cs:line 629
   at libplctag.NativeTagWrapper.GetNativeValueAndThrowOnSpecificResult[T](Func`3 nativeMethod, Int32 offset, T valueIndicatingPossibleError) in E:\Harro\libplctag.NET\src\libplctag\NativeTagWrapper.cs:line 655
   at libplctag.NativeTagWrapper.GetUInt32(Int32 offset) in E:\Harro\libplctag.NET\src\libplctag\NativeTagWrapper.cs:line 557
   at libplctag.Tag.GetUInt32(Int32 offset) in E:\Harro\libplctag.NET\src\libplctag\Tag.cs:line 443
   at libplctag.DataTypes.TagInfoPlcMapper.Decode(Tag tag, Int32 offset, Int32& elementSize) in E:\Harro\libplctag.NET\src\libplctag\DataTypes\TagInfoPlcMapper.cs:line 39
   at libplctag.DataTypes.TagInfoPlcMapper.Decode(Tag tag) in E:\Harro\libplctag.NET\src\libplctag\DataTypes\TagInfoPlcMapper.cs:line 81
   at libplctag.Tag`2.DecodeAll() in E:\Harro\libplctag.NET\src\libplctag\TagOfT.cs:line 222
   at libplctag.Tag`2.Read() in E:\Harro\libplctag.NET\src\libplctag\TagOfT.cs:line 178
   at CSharpDotNetCore.ExampleListTags.Run() in E:\Harro\libplctag.NET\src\Examples\CSharp DotNetCore\ExampleListTags.cs:line 33
   at CSharpDotNetCore.Program.Main(String[] args) in E:\Harro\libplctag.NET\src\Examples\CSharp DotNetCore\Program.cs:line 26

```

This is the code I am executing:

```
var tags = new Tag<TagInfoPlcMapper, TagInfo[]>()
{
    Name = "g_MYVAR.Info.IP_Address_PLC",
    Gateway = "10.12.5.71",
    Path = "1,0",
    PlcType = PlcType.ControlLogix,
    Protocol = Protocol.ab_eip                
};

tags.Read();

ConsoleTable
    .From(tags.Value.Select(t => new
    {
        t.Id,
        Type = $"0x{t.Type:X}",
        t.Name,
        t.Length,
        Dimensions = string.Join(',', t.Dimensions)
    }))
    .Configure(o => o.NumberAlignment = Alignment.Right)
    .Write(Format.Default);
```

Thanks in advance.

timyhac, 2024-02-06:
Using the tag info only works with the "@tags" tag name - you can't get the tag info for a single tag.

To achieve what you're looking for, you'll need to
1. [List all of the tag information for **all** tags](https://github.com/libplctag/libplctag.NET/blob/master/src/Examples/CSharp%20DotNetCore/ListUdtDefinitions.cs)
2. Go through that list and find the one that you want, and extract its type information.
3. Create another tag object for just that one.

contibru, 2024-02-06:
Yeah, I tried that, but the tag never shows on the list using that ListUdtDefinition example, but if I try get the value using the type related, it is working fine.

[image]

kyle-github, 2024-02-06:
Based on the type name, this does not look like a standard string.   Is it only 20 characters long?   If it is not a standard size, you will need to add definitions so that the library knows how to interpret it.   Also, note that on Micro8x0 PLCs you cannot fully list tags.  You cannot get UDT definitions.  At least there is no documented way that I have found.

You can look at the [wiki](https://github.com/libplctag/libplctag/wiki/Tag-String-Attributes) for more information on how to handle strings.

closed 2024-02-15 by contibru