libplctag/libplctag.NET #365: In the StringPlcMapper, Omron is not present

zN3utr4l, 2024-03-04:
In the StringPlcMapper, Omron is not present

```
using System;

namespace libplctag.DataTypes;

public class StringPlcMapper : PlcMapperBase<string>
{
    public override int? ElementSize => base.PlcType switch
    {
        PlcType.ControlLogix => 88,
        PlcType.Plc5 => 84,
        PlcType.Slc500 => 84,
        PlcType.LogixPccc => 84,
        PlcType.Micro800 => 256,
        PlcType.MicroLogix => 84,
        _ => throw new NotImplementedException(),
    };

    public override string Decode(Tag tag, int offset)
    {
        return tag.GetString(offset);
    }

    public override void Encode(Tag tag, int offset, string value)
    {
        tag.SetString(offset, value);
    }
}
```

I have the same issue as https://github.com/libplctag/libplctag.NET/issues/359

timyhac, 2024-03-04:
Thanks - do you know how strings are encoded for your device?

zN3utr4l, 2024-03-04:
No, how can I check?

kyle-github, 2024-03-04:
Use the core library test tool tag_rw to read a string tag.   If I remember correctly, strings on Omron have a counted string with a 2-byte count, no padding.  I do not know the maximum length of a string.

zN3utr4l, 2024-03-04:
I can read string from .Net wrapper, where can i find the maximum length of it?

In the PLC myString is defined as STRING[Custom Length]

kyle-github, 2024-03-04:
How is "MyIP" working?  Do you have DNS set up for the PLC?

It looks like the core library is correctly handling strings in this case.   So if there is a problem, it must be somehow in the .Net part?

zN3utr4l, 2024-03-04:
MyIp is localhost, i change it in this comment

zN3utr4l, 2024-03-04:
```
Library version 2.5.4.
Processing argument 1 "--type=uint8".
Processing argument 2 "--tag=protocol=ab-eip&gateway=127.0.0.1&path=1,0&plc=omron-njnx&name=MyString".
Processing argument 3 "--debug=4".
data[0]=0 (0)
data[1]=0 (0)
```

When the string is empty and not initialized/written by the PLC it looks like this

zN3utr4l, 2024-03-04:
```
libplctag_2.5.4_windows_x64\Release> tag_rw2 --type=string  '--tag=protocol=ab-eip&gateway=127.0.0.1&path=1,0&plc=omron-njnx&name=MyString'  --write="test"
Library version 2.5.4.
Processing argument 1 "--type=string".
Setting type to TYPE_STRING.
Processing argument 2 "--tag=protocol=ab-eip&gateway=127.0.0.1&path=1,0&plc=omron-njnx&name=MyString".
Processing argument 3 "--write=test".
Number of write args 1.
data[0]="TT"
Warning: truncating string 0, "test", to fit fixed string capacity!
Error while setting the string 0, error: PLCTAG_ERR_BAD_CONFIG!
```

This is the same error from .NET (on write)

zN3utr4l, 2024-03-04:
```
data[0]="TT"
Warning: truncating string 0, "test", to fit fixed string capacity!
2024-03-04 16:22:02.615 thread(1) tag(11) DETAIL plc_tag_set_string:3414 Starting with string te.
2024-03-04 16:22:02.615 thread(1) tag(11) WARN plc_tag_set_string:3460 string_capacity=2, string_last_offset=5, tag_size=4.
2024-03-04 16:22:02.615 thread(1) tag(11) WARN plc_tag_set_string:3472 Bad configuration? String capacity/size is larger than the tag buffer!
2024-03-04 16:22:02.615 thread(1) tag(11) DETAIL plc_tag_set_string:3574 Done with status PLCTAG_ERR_BAD_CONFIG.
Error while setting the string 0, error: PLCTAG_ERR_BAD_CONFIG!
```

This is the main error

kyle-github, 2024-03-04:
The library does not automatically extend the buffer for strings.  It probably should but it is a bit difficult to get this right.   The buffer is only 4 bytes.  Two of those are for the count word.   So there is only sufficient space for two characters.   You need to set the tag buffer to a size large enough to hold the string.  In this case, that would be 6 bytes.  [Here is the documentation](https://github.com/libplctag/libplctag/wiki/API#retrievingsetting-tag-size) for the core C API.

kyle-github, 2024-03-04:
Also, that error should probably be something about size rather than configuration.

zN3utr4l, 2024-03-05:
sorry but I don't understand, is the error an error in the library or in my configuration?

As I told you, the strings were created with [10] or [20], but I noticed that even an unlimited string gives the same problem.

What can I do to resolve

Returning to the **ElementSize**, they told me that the maximum of a string should be 256.

zN3utr4l, 2024-03-06:
@kyle-github, I tried to change the size but without success, look what I did in this comment: https://github.com/libplctag/libplctag.NET/issues/359#issuecomment-1980377953

timyhac, 2024-06-15:
@zN3utr4l - your diagnosis was correct and a new version of libplctag.NET has been published with your fix.

timyhac, 2024-07-10:
I believe this issue can be closed although no changes have been made to StringPlcMapper.

I'm not sure that the memory layout of this particular String tag is widespread enough to make a change to the StringPlcMapper. @zN3utr4l if you believe otherwise can you point to what the layout is and some Omron docs that discuss it.

zN3utr4l, 2024-07-15:
https://github.com/libplctag/libplctag.NET/issues/359#issuecomment-2169622819
In this issue the reading and writing of omron strings was done, and ElementSize is set to 256.

timyhac, 2024-07-15:
Page 2-10 of [this manual](https://assets.omron.eu/downloads/manual/en/v4/w502_nx_nj-series_instructions_reference_manual_en.pdf) says that STRING types can be a range of sizes.

In ControlLogix devices, there is a default STRING type which has a certain size (I think 88) but you can create other STRING types with other sizes.

Is 256 the size of the default STRING type for Omron NJ? Are there any manuals you can point to?

zN3utr4l, 2024-07-15:
https://docs.devicewise.com/Content/Products/GatewayDevelopersGuide/Devices/DeviceTypes/Omron/Omron-NJ-or-NX-supported-access.htm

Yes of course, they can be declared of various lengths but the maximum default size is 256.