libplctag/libplctag.NET #366: Array of UDT

arcimus, 2024-03-06:
Hello, I'm attempting to list and read tags in ControlLogix software written by an equipment manufacturer.  I am able to list tags and UDTs using your sample code, but I'm not sure how to enumerate arrays of UDTs.

Here's an example set of tags retrieved using a 3rd party tool:
[image]

Using libplctag.NET, I can determine that "B0501_COM262_Node2:I.Slot" is a UDT, and can retrieve the UDT definition that contains ".Fault" and ".Data" DINTs.  However, I'm not sure how to determine that "B0501_COM262_Node2:I.Slot" is an _array_ of UDTs.

B0501_COM262_Node2:I.Slot:
Type: 44080
Offset: 4
Metadata: 7
NumFields: 2
Size: 60

The 7 value for Metadata sticks out, but after searching around, I'm not sure what Metadata is used for.

timyhac, 2024-03-09:
There is some information about the meaning of the UDT info structure in the [source code here](https://github.com/libplctag/libplctag/blob/release/src/examples/list_tags_logix.c#L704) - you are onto something!

arcimus, 2024-03-11:
Hello and thanks for your reply.  In the source code you mentioned, I see:
```
     * N x field info entries
     *     uint16_t field_metadata - array element count or bit field number
     *     uint16_t field_type
     *     uint32_t field_offset
```
So, the 7 returned in the libplctag.NET Metadata field could be an array element count or bit field number.  How can I use libplctag.NET to determine if it's the former or the latter?

arcimus, 2024-03-25:
Hi and thanks.  The print_element_type function [here](https://github.com/libplctag/libplctag/blob/4793d92893906e80fcfc67d4c4ed87fd37d81df8/src/examples/list_tags_logix.c#L573) helped me identify tag types.

closed 2024-03-25 by arcimus