libplctag/libplctag.NET #368: Array of BOOL ErrorBadParam

arcimus, 2024-03-25:
Hello,

I'm having trouble reading values from a tag that is an array of BOOLs.  libplctag.NET(and pycomm3 and pylogix) state that its a DWORD [8, 0, 0].  This seems to be expected behavior [since BOOL arrays are not stored in the PLC as BOOL, but 32 bit words](https://github.com/dmroeder/pylogix/issues/155#issuecomment-758067987).

Using libplctag.NET, I can read a BOOL value from TagName[7] but not TagName[8] and up; ErrorBadParam.  I can read values using pycomm3 and pylogix all the way up to element 255.  I'm not sure if the code I'm using below is wrong, or of there is some other issue.

                var tagValue = new Tag<BoolPlcMapper, bool>()
                {
                    Name = "TagName[8]",
                    Gateway = "10.28.122.41",
                    Path = "1,0",
                    PlcType = PlcType.ControlLogix,
                    Protocol = Protocol.ab_eip
                };
                tagValue.Read();

kyle-github, 2024-03-25:
Yeah... BOOL arrays are frustrating.   The index into the array is by DINT, not by the individual BOOL.   So in your case, you only have 0-7 DINTs.  TagName[7] is the last DINT word in the underlying DINT array.

If you want to read the 9th bit (they start counting from zero, so bit 8 is the ninth bit), you can use a tag name of "TagName[0].8".  I _think_ that the .Net wrapper handles that.  The underlying C library does support it.

This is something that might be worth supporting directly in the library, but in order to know that you are handling a BOOL array, the user either needs to pass in a parameter or the library would need to list out all the tags and their types.

arcimus, 2024-05-03:
Thanks @kyle-github.  print_element_type() from https://github.com/libplctag/libplctag/blob/release/src/examples/list_tags_logix.c was quite helpful also.

closed 2024-05-03 by arcimus