libplctag/libplctag.NET #369: ASP.Net error

kaptindan, 2024-05-07:
Getting a error in ASP.Net web page trying to load the dll file. Its in the folder of the app. 

An exception of type 'System.DllNotFoundException' occurred in libplctag.dll but was not handled in user code
11:34:49:406	Unable to load DLL 'plctag': The specified module could not be found. (Exception from HRESULT: 0x8007007E)

timyhac, 2024-05-07:
I just created a new Blazor Application using the "Blazor Web App" template in Visual Studio, modified the Home page to get the libplctag version, and its working for me out of the box. I'm not sure which framework you are using, a search for "ASP.Net web page" takes me to a getting started article for Blazor.

[Is there any fault-finding you have already done to determine the cause of this? ](https://github.com/libplctag/libplctag.NET/blob/master/CONTRIBUTIONS.md#reporting-bugs)

The template uses Blazor / .NET 8 / ASP.NET Core.

To test whether libplctag.NET is working, I changed the home page in the projected from
```razor
@page "/"

<PageTitle>Home</PageTitle>

<h1>Hello, world!</h1>

Welcome to your new app.

```

to
```razor
@page "/"
@rendermode InteractiveServer

<PageTitle>Home</PageTitle>

<p role="status">Libplctag Version: @libplctag_version</p>

<button class="btn btn-primary" @onclick="GetLibplctagVersion">Click me</button>

@code {
    private string libplctag_version = "";

    private void GetLibplctagVersion()
    {
        var majorVersion = libplctag.LibPlcTag.VersionMajor;
        var minorVersion = libplctag.LibPlcTag.VersionMinor;
        var patchVersion = libplctag.LibPlcTag.VersionPatch;
        libplctag_version = $"v{majorVersion}.{minorVersion}.{patchVersion}";
    }
}
```

kaptindan, 2024-05-08:
My error shows up while trying to read a tag on a AB PLC

timyhac, 2024-05-08:
libplctag.NET is a wrapper around the C library [libplctag ](https://github.com/libplctag/libplctag), which we package as "plctag.dll". The exception is saying that it cannot load that dll.

A [`DllNotFoundException`](https://learn.microsoft.com/en-us/dotnet/api/system.dllnotfoundexception?view=net-8.0) will occur regardless of which C API is being invoked. Whether its reading a tag or getting the version, if the dll cannot be found either API call would trigger that exception.

To prove that, I modified the example from before to instead create a Tag and perform a Read on it.
I am not connected to a device, so the Read timed out - but it at least proves that the Dll is being loaded, and is attempting a read.

```razor
@page "/"
@rendermode InteractiveServer

<PageTitle>Home</PageTitle>

<p role="status">Read result: @result</p>

<button class="btn btn-primary" @onclick="ReadMyTag">Read MyTag</button>

@code {
    private string result = "";

    private void ReadMyTag()
    {
        var myTag = new libplctag.Tag()
            {
                Name = "MyTag",
                Gateway = "127.0.0.1",
                ElementSize = 4,
                Protocol = libplctag.Protocol.ab_eip,
                Path = "1,0",
                PlcType = libplctag.PlcType.ControlLogix,
            };

        try
        {
            myTag.Read();
            result = myTag.GetUInt32(0).ToString();
        }
        catch (Exception e)
        {
            result = e.Message;
        }
    }
}
```

[image]

Unfortunately this issue will require some additional faultfinding on your behalf to determine what the cause is. Some relevant questions from the [bug reporting guidance](https://github.com/libplctag/libplctag.NET/blob/master/CONTRIBUTIONS.md#reporting-bugs):
* Supply a [minimal reproducible example](https://stackoverflow.com/help/minimal-reproducible-example).
* What else have you tried? And why did it not work?
* Is there anything related to this that _does_ work?
* Supply [libplctag debug logs](https://github.com/libplctag/libplctag.NET/blob/master/src/Examples/CSharp%20DotNetCore/LoggingExample.cs).
* Are external conditions relevant (e.g. network latency)? Document these.

timyhac, 2024-05-08:
Going back through the issue history on this repository, it looks like you've had some success with this library in the past - so something must have changed. e.g maybe the distribution method or the framework or the version of .NET, there are many combinations of this so my test above probably wasn't that relevant to your situation - my bet is that a minimal repro will help a lot.

kaptindan, 2024-05-13:
Ok it was stupid simple, for some reason the install from NuGet did not put the plc.dll in the bin folder....once I put it there it worked.

closed 2024-06-13 by timyhac