libplctag/libplctag.NET #371: List all tags in Omron NJ: ErrorUnsupported

hamduc7, 2024-05-24:
Dear All,

I am using the latest libplctag 1.2.0 and NativeImport 1.0.37.
I want to list all tags in my device **Omron NJ** but I got the error ErrorUnsupported 

 var tags = new Tag<TagInfoPlcMapper, TagInfo[]>()
            {
                Gateway = "MY IP",
                Path = "1,0",
                PlcType = PlcType.Omron,
                Protocol = Protocol.ab_eip,
                Name = "@tags"
            };

            tags.Read();

Stack trace:
**ErrorUnsupported**
   at libplctag.NativeTagWrapper.ThrowIfStatusNotOk(Nullable`1 status)
   at libplctag.NativeTagWrapper.Read()
   at libplctag.Tag`2.Read()
   at ConsoleApp1.Program.test1()
   at ConsoleApp1.Program.Main(String[] args)

Could anyone help me to solve it?

Thank you

kyle-github, 2024-05-25:
In the core C library there is commented out code that guarded against the use of tag listing for Omron.  Based on the fact that it is not commented out, I am not sure if it actually supported with Omron PLCs.

Could you turn debugging on to level 4, capture the output, and attach it here?   Then we can tell if the PLC is telling us that tag listing is not supported or if it is something in the library.   Usually the error you see is returned by the PLC.

hamduc7, 2024-05-26:
Hi @kyle-github 
Thank you for getting back to me. Please check the attached file
[omron nj.txt](https://github.com/libplctag/libplctag.NET/files/15445691/omron.nj.txt)

I think the device does support it because there is commercial software installed on the machine and it shows all tags

By the way, can I downgrade to the latest version that still supports this function to try? If so, could you know the version?

Thank you very much.

kyle-github, 2024-05-26:
Here is the request sent to the Omron PLC:

```
55 03 20 6b 25 00 00 00 04 00 02 00 07 00 08 00 01 00
```

55 - the command to get tag information, this is getting multiple attribute of multiple objects
03 - the length of the path to the CIP object, 3 words or 6 bytes.
20 6b - Class 6b
25 00 00 00 - 16-bit instance, in this case instance 0
04 00 - we are asking for 4 attributes
02 00 07 00 08 00 01 00 - attribute 2, 7, 8 and 1

Here is the response.

```
70 00 1a 00 73 01 4a 00 00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00 00 00 00 00 00 00 02 00
a1 00 04 00 47 4f 00 00 b1 00 06 00 01 00 d5 00
08 00
```

The important part is `d5 00 08 00` at the end.

The `d5` indicates that our `55` command above completed (the upper bit is set).

The `08` means that the command was not supported or understood by the PLC.  That means that your PLC does not understand the tag listing command.  There are commands that list tags for Omron because the IDEs work :-) That said, Omron has its own set of commands and CIP objects but they are not documented anywhere I can find.

The tag listing commands _for Rockwell PLCs_ are documented by Rockwell, but they may be Rockwell-specific.   It certainly seems like it.

Without documentation or working code I can reverse engineer, listing tags looks like it is not supported on Omron.

hamduc7, 2024-05-27:
Dear @kyle-github 
Thank you very much for your detailed explanation.
Let me find the Omron NJ Ethernet document to find some more details.

Thank you very much

hamduc7, 2024-05-27:
Hi @kyle-github 
The **Path = "1,0"** is the Backplane port
Our device is Omron NJ2 Ethernet so we use an internet port. The path will be 
**Path = "2,0"**

When I use path=2,0", I get error PLCTAG_ERR_REMOTE_ERR

I have attached the log file:
[omron nj.txt](https://github.com/libplctag/libplctag.NET/files/15450794/omron.nj.txt)

NJ Document: page 250 about CIP command
[w506_nx_nj-series_cpu_unit_built-in_ethernet_ip_port_users_manual_en.pdf](https://github.com/libplctag/libplctag.NET/files/15450796/w506_nx_nj-series_cpu_unit_built-in_ethernet_ip_port_users_manual_en.pdf)

Could you pls give me any idea to make it works?

Thank you very much

kyle-github, 2024-05-27:
From what I have seen (I do not have an Omron PLC), you need to use a path like this `18,10.1.2.3`.  Change `10.1.2.3` to the IP address of your PLC.  I am not sure why but it appears that Omron PLCs route as if they are using bridging across a backplane.

hamduc7, 2024-05-27:
Thank you. Let me try now..

hamduc7, 2024-05-27:
@kyle-github I got a very long log, looks like it cannot connect when using
IP: 10.1.2.3 ->I am also not sure about this IP
Path = "1,0" (i also try "2,0")

[omron2.nj.txt](https://github.com/libplctag/libplctag.NET/files/15450981/omron2.nj.txt)

P/S: we have a commercial software keepware that can show list of tags in the device

Regards,

zN3utr4l, 2024-06-15:
> Hi @kyle-github Thank you for getting back to me. Please check the attached file:
>
> [omron nj.txt](https://github.com/libplctag/libplctag.NET/files/15445701/omron.nj.txt)
>
> I think the device does [...]

Maybe that's not your IP, as @kyle-github suggested, you need to change 10.1.2.3 for your IP, in this .txt you sent the IP 10.22.72.10 seems to be yours.

hamduc7, 2024-06-16:
@zN3utr4l 

Thank you. Using 10.1.2.3 is not connectable as I sent the log file above.
[image]

zN3utr4l, 2024-06-16:
10.22.72.10

hamduc7, 2024-06-18:
Hi @zN3utr4l 
Sorry, I don't understand what you mean. I have tried these ways but failed

1/IP: 10.22.72.10 
Path = "1,0" (i also try "2,0")

2/ IP: 10.1.2.3
Path = "1,0" (i also try "2,0")

I also tried path="18,0" but received **ErrorBadGateway**

zN3utr4l, 2024-06-18:
are you sure about that IP?

I use `ip="192.168.253.2" path="18,192.168.253.2"` and everything is fine.

kyle-github, 2024-06-18:
the paths are in the form of port/link address.   So 2,0 means "use port 2 and go to link address 0."   Port two is only available on some types of module.  It is the external port on a network module.  Since you are coming _in_ on an Ethernet connection, you are coming in on port 2.  Port 1 is the backplane.   The first two parts of a path are almost always 1,x where x is the slot in which the next module is placed.  Often the CPU is the target and often that is in slot 0.  

Most people are finding that a path of "1,0" works for Omron but the original testing showed that you often needed it to look like a bridged connection.  In that case 18,w.x.y.z seemed to work.  The "w.x.y.z" part is the IP address such as 192.168.14.3.

"18" is the following in binary:

001 - logical segment
00010 - link/port 2

So port two.  The logical segment means a string, in this case the IP address.

As @zN3utr4l said, check your IP addresses.

hamduc7, 2024-06-19:
@kyle-github @zN3utr4l Thank you very much for your detailed instructions. Let me study more.

hamduc7, 2024-07-01:
@kyle-github @zN3utr4l 
The IP and Path are correct because **I can read a float tag value successfully**.

It only does not work when I want to list tags using @tags.

` var tags = new Tag<TagInfoPlcMapper, TagInfo[]>()
                {
                    Gateway = "10.22.72.10",
                    Path = "1,0",
                    PlcType = PlcType.Omron,
                    Protocol = Protocol.ab_eip,
                    Name = "@tags",
                    DebugLevel = DebugLevel.Detail,
                    Timeout = TimeSpan.FromSeconds(1000)
                };

                tags.Read();`
 
 
[log.txt](https://github.com/user-attachments/files/16050231/log.txt)

Could you please provide some more hints?

Thank you

zN3utr4l, 2024-07-01:
Item navigation is not supported on Omron

hamduc7, 2024-07-01:
@zN3utr4l There is a commercial software, it can show all tags on my device. So it is possible

zN3utr4l, 2024-07-03:
https://github.com/libplctag/libplctag.NET/issues/359#issuecomment-1991647955 look here. [aphytcomm](https://github.com/aphyt/aphytcomm)

hamduc7, 2024-07-04:
Hi all,
The last codes in the below script solves my issue
https://github.com/libplctag/libplctag/issues/466#issuecomment-2200324984

Thank you very much for support from @MountainKing91

closed 2024-07-04 by hamduc7