libplctag/libplctag.NET #373: How to batch read multiple tags at once?

313843288, 2024-06-03:
AB PLC，How to batch read multiple tags at once？

kyle-github, 2024-06-03:
If the PLC supports it, the library does this automatically if it has multiple tags with outstanding requests.  The simplest way to make sure that this happens is to create several tags and then trigger reads or writes on all of them without a wait time.  This triggers the operation in the background.  Once they are all triggered, then just wait on all of the tags.

kyle-github, 2024-06-03:
I am not a C# programmer, but I think the code at the [indicated section](https://github.com/libplctag/libplctag.NET/blob/e92962ffc2d51951140703576ade44b22d9a8264/src/LibplctagReadBenchmark/ReadExamples.cs#L21) does the async read of multiple tags.

closed 2024-06-04 by 313843288