libplctag/libplctag.NET #380: DotNetFramework can not work

njsunsky, 2024-06-17:
The exception thrown: "System.IO.FileNotFoundException" (occurred in libplctag.dll)

An unhandled exception of type 'System.IO.FileNotFoundException' occurred in libplctag.dll
Could not load file or assembly 'libplctag.NativeImport, Version=1.0.39.0, Culture=neutral, PublicKeyToken=null' or one of its dependencies. The system cannot find the file specified.

timyhac, 2024-06-17:
I'm not sure when this started happening but it seems like the transitive dependency libplctag.NativeImport isn't being brought in when libplctag.NET (primary .NET wrapper package) is project reference as opposed to a nuget package reference.

You should be able to get around the issue by installing either of the nuget packages.

njsunsky, 2024-06-22:
@timyhac thanks ,you are right ,i install  either of the nuget packages ,it works well

closed 2024-06-22 by njsunsky