libplctag/libplctag.NET #381: Read array tags for Omron PLC broken with v1.4.0

MountainKing91, 2024-06-17:
After upgrading from libplctag.NET 1.20. to 1.4.0 it is not possible to read arrays

PLC: Omron NX1P2

For example, I want to read a tag named "AlarmList", datatype BOOL[64]

Code:
```
var tag = new Tag<BoolPlcMapper, bool[]>()
{
    Name = "AlarmList",
    Gateway = "192.168.250.1",
    Path = "1,0",
    PlcType = PlcType.Omron,
    Protocol = Protocol.ab_eip,
    ArrayDimensions = new int[] { 64 },
    Timeout = TimeSpan.FromMilliseconds(5000)
};

tag.Read();
```

This code works fine with version 1.2.0 but throws an exception "ErrorBadParam" with 1.4.0. I looked at the changelog but I couldn't locate what has changed regarding arrays.

kyle-github, 2024-06-17:
Bad Param sounds like a core library error.  Can you capture the logs and attach them?  Use debug level 4 (detail).  If the version of the .Net wrapper you are using uses version 2.6.0 of the core library, then that has a lot of small changes for Omron contributed by a user.

MountainKing91, 2024-06-17:
> Bad Param sounds like a core library error. Can you capture the logs and attach them? Use debug level 4 (detail). If the version of the .Net wrapper you are using uses version 2.6.0 of the core library, then that has a lot of small changes for Omron contributed by a user.

Attached in my opening comment

kyle-github, 2024-06-17:
D'oh.  I shouldn't comment on my phone...

kyle-github, 2024-06-17:
Huh... It is the PLC that is sending that back.   The error is major code 0x20 and minor code 0x8017.  In the "fine" manual from Omron, it says that this is due to:

```
More than one element was specified for a variable that does not have elements.
```

Hmmm.  Let's look at the request.

```
4c 06 91 09 41 6c 61 72 6d 4c 69 73 74 00 02 00
```

4c - read command
06 - number of words in the tag path.
91 09 41 6c 61 72 6d 4c 69 73 74 00 - tag name with padding byte because the name is an odd length.
02 00 - get two elements

I think this worked before because the core library was overriding the element count setting it to 1 all the time.   Now the element count is 2 but there isn't an array element specifier on the tag path.  It is just a bare name.  Omron's manual says that when an array is accessed with just the bare name, all the elements are returned.  So asking for 2 is really asking for the array and a second array.   At least that is how I read the docs.  Omron's manual is... not great.   It is actually worse than Rockwell's docs.

Try two things:

1. Set the element count to 1.  If it works, you should get the whole array.  Possibly packed as bits and possibly as some larger unit per BOOL.
2. Use a tag path of "AlarmList[0]" and leave the element count to 2.

I will be very interested to see if the response in 2 will be two words (D2 type IIRC).

kyle-github, 2024-06-17:
So where is that count of two elements coming from?   It is not clear in the code you showed above.

kyle-github, 2024-06-17:
Ahh. I think I see.  The array dimensions are given as a single dimension of 64.  If this was a Rockwell PLC it would use 32-bit double words to store the bits.  IIRC, Omron uses 16-bit words.  So 64/32 = 2 for Rockwell, but would be 4 words for Omron.  But since this is a bare tag name, the entire array is handed back at once.  I think that option 2 above will probably work and generate something.  Whether it is what you wanted is a different story.

Try another option:

3. Change the dimension to 32 from 64 and see if that loads something.  Omron may still be OK with an element count as long as it is one.

MountainKing91, 2024-06-17:
I was going to ask exactly that; I posted all my code, there is nothing else.

I tried what you suggested:

1. It works for ArrayDimensions ranging from { 1 } to { 32 }, but I get only 32 bool values
2. With name "AlarmList[0]" I get an ErrorOutOfBounds exception no matter what other changes I make
3. See point 1

kyle-github, 2024-06-17:
Can you post debugging output for the three cases.  I want to see if 1 and 3 are bringing back the same data and whether the error in 2 is from the core DLL or the PLC.

Thanks.

MountainKing91, 2024-06-17:
Sure!

Case 1, array dimension { 1 }, tag name AlarmList
```
2024-06-17 21:11:34.650 thread(1) tag(0) INFO plc_tag_create_ex:838 Starting
2024-06-17 21:11:34.652 thread(1) tag(0) INFO initialize_modules:177 Starting.
2024-06-17 21:11:34.652 thread(1) tag(0) INFO initialize_modules:186 Creating library mutex.
2024-06-17 21:11:34.652 thread(1) tag(0) DETAIL mutex_create:726 Starting.
2024-06-17 21:11:34.652 thread(1) tag(0) DETAIL mutex_create:753 Done.
2024-06-17 21:11:34.652 thread(1) tag(0) INFO initialize_modules:207 Initializing library modules.
2024-06-17 21:11:34.652 thread(1) tag(0) INFO lib_init:136 Starting.
2024-06-17 21:11:34.652 thread(1) tag(0) INFO lib_init:140 Setting up global library data.
2024-06-17 21:11:34.652 thread(1) tag(0) INFO lib_init:142 Creating tag hashtable.
2024-06-17 21:11:34.652 thread(1) tag(0) INFO hashtable_create:75 Starting
2024-06-17 21:11:34.652 thread(1) tag(0) INFO hashtable_create:99 Done
2024-06-17 21:11:34.652 thread(1) tag(0) INFO lib_init:148 Creating tag hashtable mutex.
2024-06-17 21:11:34.652 thread(1) tag(0) DETAIL mutex_create:726 Starting.
2024-06-17 21:11:34.652 thread(1) tag(0) DETAIL mutex_create:753 Done.
2024-06-17 21:11:34.652 thread(1) tag(0) INFO lib_init:154 Creating tag condition variable.
2024-06-17 21:11:34.652 thread(1) tag(0) DETAIL cond_create:1087 Starting.
2024-06-17 21:11:34.652 thread(1) tag(0) DETAIL cond_create:1114 Done.
2024-06-17 21:11:34.652 thread(1) tag(0) INFO lib_init:160 Creating tag tickler thread.
2024-06-17 21:11:34.652 thread(1) tag(0) DETAIL thread_create:884 Starting.
2024-06-17 21:11:34.652 thread(1) tag(0) DETAIL thread_create:917 Done.
2024-06-17 21:11:34.652 thread(1) tag(0) INFO lib_init:166 Done.
2024-06-17 21:11:34.652 thread(1) tag(0) INFO initialize_modules:210 Initializing AB module.
2024-06-17 21:11:34.652 thread(1) tag(0) INFO ab_init:125 Initializing AB protocol library.
2024-06-17 21:11:34.652 thread(2) tag(0) INFO tag_tickler_func:513 Starting.
2024-06-17 21:11:34.652 thread(1) tag(0) DETAIL mutex_create:726 Starting.
2024-06-17 21:11:34.652 thread(1) tag(0) DETAIL mutex_create:753 Done.
2024-06-17 21:11:34.652 thread(1) tag(0) INFO ab_init:134 Finished initializing AB protocol library.
2024-06-17 21:11:34.652 thread(1) tag(0) INFO initialize_modules:215 Initializing Modbus module.
2024-06-17 21:11:34.652 thread(1) tag(0) INFO mb_init:2469 Starting.
2024-06-17 21:11:34.652 thread(1) tag(0) DETAIL mb_init:2471 Setting up mutex.
2024-06-17 21:11:34.652 thread(1) tag(0) DETAIL mutex_create:726 Starting.
2024-06-17 21:11:34.652 thread(1) tag(0) DETAIL mutex_create:753 Done.
2024-06-17 21:11:34.652 thread(1) tag(0) INFO mb_init:2480 Done.
2024-06-17 21:11:34.652 thread(1) tag(0) INFO initialize_modules:226 Done initializing library modules.
2024-06-17 21:11:34.652 thread(1) tag(0) INFO initialize_modules:231 Done.
2024-06-17 21:11:34.652 thread(1) tag(0) DETAIL attr_create_from_str:129 Starting.
2024-06-17 21:11:34.654 thread(1) tag(0) DETAIL attr_create_from_str:158 Key-value pair "protocol=ab_eip".
2024-06-17 21:11:34.654 thread(1) tag(0) DETAIL attr_create_from_str:173 Key-value pair before trimming "protocol":"ab_eip".
2024-06-17 21:11:34.654 thread(1) tag(0) DETAIL attr_create_from_str:185 Key-value pair after trimming "protocol":"ab_eip".
2024-06-17 21:11:34.654 thread(1) tag(0) DETAIL attr_create_from_str:158 Key-value pair "gateway=192.168.250.1".
2024-06-17 21:11:34.654 thread(1) tag(0) DETAIL attr_create_from_str:173 Key-value pair before trimming "gateway":"192.168.250.1".
2024-06-17 21:11:34.654 thread(1) tag(0) DETAIL attr_create_from_str:185 Key-value pair after trimming "gateway":"192.168.250.1".
2024-06-17 21:11:34.654 thread(1) tag(0) DETAIL attr_create_from_str:158 Key-value pair "path=1,0".
2024-06-17 21:11:34.654 thread(1) tag(0) DETAIL attr_create_from_str:173 Key-value pair before trimming "path":"1,0".
2024-06-17 21:11:34.654 thread(1) tag(0) DETAIL attr_create_from_str:185 Key-value pair after trimming "path":"1,0".
2024-06-17 21:11:34.654 thread(1) tag(0) DETAIL attr_create_from_str:158 Key-value pair "plc=omron-njnx".
2024-06-17 21:11:34.654 thread(1) tag(0) DETAIL attr_create_from_str:173 Key-value pair before trimming "plc":"omron-njnx".
2024-06-17 21:11:34.654 thread(1) tag(0) DETAIL attr_create_from_str:185 Key-value pair after trimming "plc":"omron-njnx".
2024-06-17 21:11:34.654 thread(1) tag(0) DETAIL attr_create_from_str:158 Key-value pair "elem_size=1".
2024-06-17 21:11:34.654 thread(1) tag(0) DETAIL attr_create_from_str:173 Key-value pair before trimming "elem_size":"1".
2024-06-17 21:11:34.654 thread(1) tag(0) DETAIL attr_create_from_str:185 Key-value pair after trimming "elem_size":"1".
2024-06-17 21:11:34.656 thread(1) tag(0) DETAIL attr_create_from_str:158 Key-value pair "elem_count=1".
2024-06-17 21:11:34.656 thread(1) tag(0) DETAIL attr_create_from_str:173 Key-value pair before trimming "elem_count":"1".
2024-06-17 21:11:34.656 thread(1) tag(0) DETAIL attr_create_from_str:185 Key-value pair after trimming "elem_count":"1".
2024-06-17 21:11:34.656 thread(1) tag(0) DETAIL attr_create_from_str:158 Key-value pair "name=AlarmList".
2024-06-17 21:11:34.656 thread(1) tag(0) DETAIL attr_create_from_str:173 Key-value pair before trimming "name":"AlarmList".
2024-06-17 21:11:34.656 thread(1) tag(0) DETAIL attr_create_from_str:185 Key-value pair after trimming "name":"AlarmList".
2024-06-17 21:11:34.656 thread(1) tag(0) DETAIL attr_create_from_str:216 Done.
[... cut ...]
2024-06-17 21:17:28.162 thread(2) tag(11) DETAIL tag_tickler_func:550 Tickling tag 11.
2024-06-17 21:17:28.164 thread(1) tag(0) INFO plc_tag_read:1737 Done
2024-06-17 21:17:28.164 thread(2) tag(11) DETAIL plc_tag_generic_tickler:281 Tickling tag 11.
2024-06-17 21:17:28.164 thread(2) tag(11) DETAIL plc_tag_generic_tickler:386 Done.
32
True,False,True,False,False,False,True,True,False,False,False,False,False,False,False,False,False,False,False,False,False,False,False,False,False,False,False,False,False,False,False,False
```

kyle-github, 2024-06-17:
Phwew.   Thanks for all that.

Here's my analysis of this:

First test payload:
Case 1, array dimension { 1 }, tag name AlarmList

Request: 4c 06 91 09 41 6c 61 72 6d 4c 69 73 74 00 01 00
One element requested

Response: cc 00 00 00 c1 00 c5 00 00 00 00 00 00 00
Type 0x00c1 = BIT, but we get 8 bytes/64 bits of data.  That's the whole array.  It is clear that bits are packed here.

Case 2, ArrayDimension { 2 }, tag name AlarmList[0]

Request: 4c 07 91 09 41 6c 61 72 6d 4c 69 73 74 00 28 00 01 00
the 28 00 part is index 0 in the array [0].  Still requesting one element.

Response: cc 00 00 00 c1 00 01 00
Same type 0x00c1 but now we only got two bytes.  Maybe each byte is one of the bits, since two were requested?  We had an index so we did not get the whole tag.

This gets an out of bounds error in the plc_tag_get_bit() function which means we requested either something negative or a bit index greater than 15.  Unfortunately, I don't print out the index so I cannot tell here.

Case 3, same as 1 but with array dimension 32

Request: 4c 06 91 09 41 6c 61 72 6d 4c 69 73 74 00 01 00
Same request as case 1.  So we get the whole tag.

Response: cc 00 00 00 c1 00 c5 00 00 00 00 00 00 00
And we get the same data. 64-bits worth.

Here is what I think is going on.  I think that the wrapper is assuming that BOOL arrays are allocated in chunks of 32 bits (that's what Rockwell does).  The wrapper assumed you will get a 32-bit bit string back even if you request a single bit/BOOL.  So in case 2, it is trying to read all the bits out but there are only two bytes.

Here's a query for my ControlLogix where I have a BOOL array, TestBOOLArray.  I asked for the tag name alone and 2 elements:

Request: 52 08 91 0d 54 65 73 74 42 4f 4f 4c 41 72 72 61 79 00 02 00 00 00 00 00

I used the fragmented read form, so the last four bytes are the byte offset, in this case zero.   

Response: d2 00 00 00 d3 00 02 00 00 00 00 00 00 00

Here we get type D3 back which is a 32-bit bitstring.  And I get two of them.  So the element count is the number of backing bit string integers.

The return of the type D3 is a really good flag to tell us that we are dealing with a bit string/BOOL array.  

For Omron, in case 2 it is not clear that we got two bytes because we asked for two elements or if we got two bytes because that is the size of the backing integer for a BOOL array (it is 16-bits in Omron PLCs).  The fact that we got a single 1 in the bytes indicates to me that each byte might be representing a single bit.  Or maybe the other bits are masked off?  

Can you run test 2 again, but ask for 8 elements?  We have a couple likely possibilities:

A: We still get two bytes back because they are packed and Omron is masking off the bits we did not ask for.  The first byte will have the value 0xC5 in it.  The bit indexing operation will still fail.

B: We will get 8 bytes back and and they will look like this:

01 00 01 00 00 00 01 01

In this case the bit indexing operation will succeed.

kyle-github, 2024-06-17:
It is very annoying that Omron does not give us some information about the tag type through the element type information.  Rockwell uses the bit-string types when they return a BOOL array.  Omron just uses the usual single-bit type.

MountainKing91, 2024-06-17:
What omron resources do you have to look up?

timyhac, 2024-06-17:
It may be useful in this case to use the [`Tag`](https://github.com/libplctag/libplctag.NET/blob/master/src/libplctag/Tag.cs) class (not `Tag<M,T>`) to avoid running into issues with the Mapper system. You're quite right that that the BoolMapper was designed for Rockwell.
It is possible for the Mapper to have different behaviour depending on the target device, but since we're not sure what the correct behaviour is it might be more productive to avoid it altogether and use the underlying Tag directly.

kyle-github, 2024-06-17:
[w506_nx_nj-series_cpu_unit_built-in_ethernet_ip_port_users_manual_en.pdf](https://github.com/user-attachments/files/15877453/w506_nx_nj-series_cpu_unit_built-in_ethernet_ip_port_users_manual_en.pdf)

See attached.  That's all I've been able to find.  That and reading through the Aphytcomm source code.  

So few of the details of all the CIP-based PLCs are documented well that it is very common to dive into the bytes and Wireshark captures to figure out what is going on.   Rockwell seems to be a bit better at documenting things than Omron, but not much better.

I have been trying to avoid reading all the tag definitions right at the beginning of a session with a PLC.  If you have a PLC program with hundreds (or thousands) of tags, that process can take a long, long time.  Many seconds.  That is unacceptable start up time for many applications.  And it uses a lot of memory. 

The base C DLL is designed to be fairly small and is in use in several places on relatively small hardware.  I am working on changes to make the memory use even smaller but those are not ready yet.  Having the entire tag data base in memory all the time is far too much memory for these limited systems.

Omron does have one interesting difference from Rockwell.  It looks like you can send the Get_Attributes_All CIP service call to a tag name directly.  Can't do that on Rockwell.

MountainKing91, 2024-06-18:
````> Can you run test 2 again, but ask for 8 elements? We have a couple likely possibilities:
> 
> A: We still get two bytes back because they are packed and Omron is masking off the bits we did not ask for. The first byte will have the value 0xC5 in it. The bit indexing operation will still fail.
> 
> B: We will get 8 bytes back and and they will look like this:
> 
> 01 00 01 00 00 00 01 01
> 
> In this case the bit indexing operation will succeed.

With tag name AlarmList[0] and array dimension { 8 } I get ErrorOutOfBounds
```
2024-06-18 10:30:18.587 thread(1) tag(0) INFO plc_tag_create_ex:838 Starting
2024-06-18 10:30:18.589 thread(1) tag(0) INFO initialize_modules:177 Starting.
2024-06-18 10:30:18.590 thread(1) tag(0) INFO initialize_modules:186 Creating library mutex.
2024-06-18 10:30:18.590 thread(1) tag(0) DETAIL mutex_create:726 Starting.
2024-06-18 10:30:18.590 thread(1) tag(0) DETAIL mutex_create:753 Done.
2024-06-18 10:30:18.590 thread(1) tag(0) INFO initialize_modules:207 Initializing library modules.
2024-06-18 10:30:18.590 thread(1) tag(0) INFO lib_init:136 Starting.
2024-06-18 10:30:18.590 thread(1) tag(0) INFO lib_init:140 Setting up global library data.
2024-06-18 10:30:18.590 thread(1) tag(0) INFO lib_init:142 Creating tag hashtable.
2024-06-18 10:30:18.590 thread(1) tag(0) INFO hashtable_create:75 Starting
2024-06-18 10:30:18.590 thread(1) tag(0) INFO hashtable_create:99 Done
2024-06-18 10:30:18.590 thread(1) tag(0) INFO lib_init:148 Creating tag hashtable mutex.
2024-06-18 10:30:18.590 thread(1) tag(0) DETAIL mutex_create:726 Starting.
2024-06-18 10:30:18.590 thread(1) tag(0) DETAIL mutex_create:753 Done.
2024-06-18 10:30:18.590 thread(1) tag(0) INFO lib_init:154 Creating tag condition variable.
2024-06-18 10:30:18.590 thread(1) tag(0) DETAIL cond_create:1087 Starting.
2024-06-18 10:30:18.590 thread(1) tag(0) DETAIL cond_create:1114 Done.
2024-06-18 10:30:18.590 thread(1) tag(0) INFO lib_init:160 Creating tag tickler thread.
2024-06-18 10:30:18.590 thread(1) tag(0) DETAIL thread_create:884 Starting.
2024-06-18 10:30:18.590 thread(1) tag(0) DETAIL thread_create:917 Done.
2024-06-18 10:30:18.590 thread(1) tag(0) INFO lib_init:166 Done.
2024-06-18 10:30:18.590 thread(1) tag(0) INFO initialize_modules:210 Initializing AB module.
2024-06-18 10:30:18.590 thread(1) tag(0) INFO ab_init:125 Initializing AB protocol library.
2024-06-18 10:30:18.590 thread(1) tag(0) DETAIL mutex_create:726 Starting.
2024-06-18 10:30:18.590 thread(1) tag(0) DETAIL mutex_create:753 Done.
2024-06-18 10:30:18.590 thread(1) tag(0) INFO ab_init:134 Finished initializing AB protocol library.
2024-06-18 10:30:18.590 thread(2) tag(0) INFO tag_tickler_func:513 Starting.
2024-06-18 10:30:18.590 thread(1) tag(0) INFO initialize_modules:215 Initializing Modbus module.
2024-06-18 10:30:18.590 thread(1) tag(0) INFO mb_init:2469 Starting.
2024-06-18 10:30:18.590 thread(1) tag(0) DETAIL mb_init:2471 Setting up mutex.
2024-06-18 10:30:18.590 thread(1) tag(0) DETAIL mutex_create:726 Starting.
2024-06-18 10:30:18.590 thread(1) tag(0) DETAIL mutex_create:753 Done.
2024-06-18 10:30:18.590 thread(1) tag(0) INFO mb_init:2480 Done.
2024-06-18 10:30:18.590 thread(1) tag(0) INFO initialize_modules:226 Done initializing library modules.
2024-06-18 10:30:18.590 thread(1) tag(0) INFO initialize_modules:231 Done.
2024-06-18 10:30:18.590 thread(1) tag(0) DETAIL attr_create_from_str:129 Starting.
2024-06-18 10:30:18.590 thread(1) tag(0) DETAIL attr_create_from_str:158 Key-value pair "protocol=ab_eip".
2024-06-18 10:30:18.590 thread(1) tag(0) DETAIL attr_create_from_str:173 Key-value pair before trimming "protocol":"ab_eip".
2024-06-18 10:30:18.590 thread(1) tag(0) DETAIL attr_create_from_str:185 Key-value pair after trimming "protocol":"ab_eip".
2024-06-18 10:30:18.592 thread(1) tag(0) DETAIL attr_create_from_str:158 Key-value pair "gateway=192.168.250.1".
2024-06-18 10:30:18.592 thread(1) tag(0) DETAIL attr_create_from_str:173 Key-value pair before trimming "gateway":"192.168.250.1".
2024-06-18 10:30:18.592 thread(1) tag(0) DETAIL attr_create_from_str:185 Key-value pair after trimming "gateway":"192.168.250.1".
2024-06-18 10:30:18.592 thread(1) tag(0) DETAIL attr_create_from_str:158 Key-value pair "path=1,0".
2024-06-18 10:30:18.592 thread(1) tag(0) DETAIL attr_create_from_str:173 Key-value pair before trimming "path":"1,0".
2024-06-18 10:30:18.592 thread(1) tag(0) DETAIL attr_create_from_str:185 Key-value pair after trimming "path":"1,0".
2024-06-18 10:30:18.592 thread(1) tag(0) DETAIL attr_create_from_str:158 Key-value pair "plc=omron-njnx".
2024-06-18 10:30:18.592 thread(1) tag(0) DETAIL attr_create_from_str:173 Key-value pair before trimming "plc":"omron-njnx".
2024-06-18 10:30:18.592 thread(1) tag(0) DETAIL attr_create_from_str:185 Key-value pair after trimming "plc":"omron-njnx".
2024-06-18 10:30:18.592 thread(1) tag(0) DETAIL attr_create_from_str:158 Key-value pair "elem_size=1".
2024-06-18 10:30:18.592 thread(1) tag(0) DETAIL attr_create_from_str:173 Key-value pair before trimming "elem_size":"1".
2024-06-18 10:30:18.592 thread(1) tag(0) DETAIL attr_create_from_str:185 Key-value pair after trimming "elem_size":"1".
2024-06-18 10:30:18.592 thread(1) tag(0) DETAIL attr_create_from_str:158 Key-value pair "elem_count=1".
2024-06-18 10:30:18.592 thread(1) tag(0) DETAIL attr_create_from_str:173 Key-value pair before trimming "elem_count":"1".
[... cut ...]
2024-06-18 10:30:18.671 thread(1) tag(11) INFO plc_tag_read:1724 elapsed time 18ms
2024-06-18 10:30:18.671 thread(1) tag(0) INFO plc_tag_read:1737 Done
2024-06-18 10:30:18.671 thread(2) tag(11) DETAIL tag_tickler_func:550 Tickling tag 11.
2024-06-18 10:30:18.671 thread(2) tag(11) DETAIL plc_tag_generic_tickler:281 Tickling tag 11.
2024-06-18 10:30:18.673 thread(2) tag(11) DETAIL plc_tag_generic_tickler:386 Done.
2024-06-18 10:30:18.673 thread(1) tag(11) WARN plc_tag_get_bit:2280 Data offset out of bounds!
ErrorOutOfBounds
```

MountainKing91, 2024-06-18:
@kyle-github
> For Omron, in case 2 it is not clear that we got two bytes because we asked for two elements or if we got two bytes because that is the size of the backing integer for a BOOL array (it is 16-bits in [...]

Could this be clarified looking at a different array? For example DINT[64] or something else.

timyhac, 2024-06-18:
Doing this test with a DINT array will at least help to know that the problem is localised to BOOL arrays.

The Mapper that you were using for BOOL arrays sets ElementSize to 1 (byte), and assumes [ElementCount is packed into a 32 bit double word](https://github.com/libplctag/libplctag.NET/blob/master/src/libplctag/DataTypes/BoolPlcMapper.cs#L22) so in this case would set ElementCount to 2 - this was developed Rockwell ControlLogix PLCs (although I'm not sure how appropriate even that is that is).

I haven't been deeply reading this thread but it seems like you're moving towards using an ElementSize of 2 and ElementCount of 64 / 16 = 4 seems like a reasonably logical value to use.

As far as I understand, ElementSize is optional for Rockwell so the fact that I was setting it to 1 might be incorrect but unused - not entirely sure.

kyle-github, 2024-06-18:
The request shows that we are still asking for one element.

@timyhac so the mapper itself is overriding the element size?

@MountainKing91 can you try again, but use an array size of 257 (that will map to 9 elements)?  We should still see the out of bounds error, but we should see if we get more bits masked or not.

MountainKing91, 2024-06-18:
I'm getting a little confused here. The reason why I was trying to go with mappers is that I have other UDTs array to read/write, starting simple with bools _seemed_ a good idea :/

Unfortunately ElementSize = 2 and ElementCount = 4 throws again ErrorBadParam.
```
const int ARRAY_LENGTH = 4;
const int TIMEOUT = 1000;

var myArrayTag = new Tag()
{
    Name = "AlarmList",
    Gateway = "192.168.250.1",
    Path = "1,0",
    PlcType = PlcType.Omron,
    Protocol = Protocol.ab_eip,
    ElementSize = 2,
    ElementCount = ARRAY_LENGTH,
    Timeout = TimeSpan.FromMilliseconds(TIMEOUT),
};
```

timyhac, 2024-06-18:
@kyle-github the .NET package offers two Tag classes:

1. [`Tag`](https://github.com/libplctag/libplctag.NET/blob/master/src/libplctag/Tag.cs) which doesn't do much more than wrap the libplctag core behaviour. Using this class you need to set ElementSize and ElementCount, and any other attributes you want to configure. To access the data you need to use the getter/setter methods.

2. [`Tag<M,T>`](https://github.com/libplctag/libplctag.NET/blob/master/src/libplctag/TagOfT.cs) on the other hand additionally provides a system to map the PLC data into a C# type (`T`) and vice-versa. This logic is handled by a Mapper class (`M`). When using `Tag<M,T>` you have a property called ArrayDimensions instead of ElementCount, where you set the array dimensions of the tag. It supports 1d, 2d or 3d arrays. You don't set ElementCount directly and leave it up to the Mapper to figure out what the value is. The Mapper talks to the underlying `Tag` and sets ElementCount.
For most Mappers this is just multiplying the array dimensions, but for BOOL arrays it does the above magic.

kyle-github, 2024-06-18:
That's odd.  You are not using the mapper at all so nothing should be changing the element count out from underneath you...  Can you try that with the tag set to "AlarmList[0]" and capture the debug output?

MountainKing91, 2024-06-18:
@kyle-github

> can you try again, but use an array size of 257 (that will map to 9 elements)? We should still see the out of bounds error, but we should see if we get more bits masked or not.
```
const int ARRAY_LENGTH = 257;
const int TIMEOUT = 1000;

var myArrayTag = new Tag()
{
    Name = "AlarmList",
    Gateway = "192.168.250.1",
    Path = "1,0",
    PlcType = PlcType.Omron,
    Protocol = Protocol.ab_eip,
    ElementSize = 2,
    ElementCount = ARRAY_LENGTH,
    Timeout = TimeSpan.FromMilliseconds(TIMEOUT),
};
```
```
2024-06-18 12:23:21.833 thread(1) tag(0) INFO plc_tag_create_ex:838 Starting
2024-06-18 12:23:21.835 thread(1) tag(0) INFO initialize_modules:177 Starting.
2024-06-18 12:23:21.835 thread(1) tag(0) INFO initialize_modules:186 Creating library mutex.
2024-06-18 12:23:21.835 thread(1) tag(0) DETAIL mutex_create:726 Starting.
2024-06-18 12:23:21.835 thread(1) tag(0) DETAIL mutex_create:753 Done.
2024-06-18 12:23:21.835 thread(1) tag(0) INFO initialize_modules:207 Initializing library modules.
2024-06-18 12:23:21.835 thread(1) tag(0) INFO lib_init:136 Starting.
2024-06-18 12:23:21.835 thread(1) tag(0) INFO lib_init:140 Setting up global library data.
2024-06-18 12:23:21.835 thread(1) tag(0) INFO lib_init:142 Creating tag hashtable.
2024-06-18 12:23:21.835 thread(1) tag(0) INFO hashtable_create:75 Starting
2024-06-18 12:23:21.835 thread(1) tag(0) INFO hashtable_create:99 Done
2024-06-18 12:23:21.835 thread(1) tag(0) INFO lib_init:148 Creating tag hashtable mutex.
2024-06-18 12:23:21.835 thread(1) tag(0) DETAIL mutex_create:726 Starting.
2024-06-18 12:23:21.835 thread(1) tag(0) DETAIL mutex_create:753 Done.
2024-06-18 12:23:21.835 thread(1) tag(0) INFO lib_init:154 Creating tag condition variable.
2024-06-18 12:23:21.835 thread(1) tag(0) DETAIL cond_create:1087 Starting.
2024-06-18 12:23:21.835 thread(1) tag(0) DETAIL cond_create:1114 Done.
2024-06-18 12:23:21.835 thread(1) tag(0) INFO lib_init:160 Creating tag tickler thread.
2024-06-18 12:23:21.835 thread(1) tag(0) DETAIL thread_create:884 Starting.
2024-06-18 12:23:21.835 thread(1) tag(0) DETAIL thread_create:917 Done.
2024-06-18 12:23:21.835 thread(1) tag(0) INFO lib_init:166 Done.
2024-06-18 12:23:21.835 thread(1) tag(0) INFO initialize_modules:210 Initializing AB module.
2024-06-18 12:23:21.835 thread(1) tag(0) INFO ab_init:125 Initializing AB protocol library.
2024-06-18 12:23:21.835 thread(1) tag(0) DETAIL mutex_create:726 Starting.
2024-06-18 12:23:21.835 thread(1) tag(0) DETAIL mutex_create:753 Done.
2024-06-18 12:23:21.835 thread(1) tag(0) INFO ab_init:134 Finished initializing AB protocol library.
2024-06-18 12:23:21.835 thread(1) tag(0) INFO initialize_modules:215 Initializing Modbus module.
2024-06-18 12:23:21.835 thread(2) tag(0) INFO tag_tickler_func:513 Starting.
2024-06-18 12:23:21.835 thread(1) tag(0) INFO mb_init:2469 Starting.
2024-06-18 12:23:21.835 thread(1) tag(0) DETAIL mb_init:2471 Setting up mutex.
2024-06-18 12:23:21.835 thread(1) tag(0) DETAIL mutex_create:726 Starting.
2024-06-18 12:23:21.835 thread(1) tag(0) DETAIL mutex_create:753 Done.
2024-06-18 12:23:21.837 thread(1) tag(0) INFO mb_init:2480 Done.
2024-06-18 12:23:21.837 thread(1) tag(0) INFO initialize_modules:226 Done initializing library modules.
2024-06-18 12:23:21.837 thread(1) tag(0) INFO initialize_modules:231 Done.
2024-06-18 12:23:21.837 thread(1) tag(0) DETAIL attr_create_from_str:129 Starting.
2024-06-18 12:23:21.837 thread(1) tag(0) DETAIL attr_create_from_str:158 Key-value pair "protocol=ab_eip".
2024-06-18 12:23:21.837 thread(1) tag(0) DETAIL attr_create_from_str:173 Key-value pair before trimming "protocol":"ab_eip".
2024-06-18 12:23:21.837 thread(1) tag(0) DETAIL attr_create_from_str:185 Key-value pair after trimming "protocol":"ab_eip".
2024-06-18 12:23:21.837 thread(1) tag(0) DETAIL attr_create_from_str:158 Key-value pair "gateway=192.168.250.1".
2024-06-18 12:23:21.837 thread(1) tag(0) DETAIL attr_create_from_str:173 Key-value pair before trimming "gateway":"192.168.250.1".
2024-06-18 12:23:21.837 thread(1) tag(0) DETAIL attr_create_from_str:185 Key-value pair after trimming "gateway":"192.168.250.1".
2024-06-18 12:23:21.837 thread(1) tag(0) DETAIL attr_create_from_str:158 Key-value pair "path=1,0".
2024-06-18 12:23:21.837 thread(1) tag(0) DETAIL attr_create_from_str:173 Key-value pair before trimming "path":"1,0".
2024-06-18 12:23:21.837 thread(1) tag(0) DETAIL attr_create_from_str:185 Key-value pair after trimming "path":"1,0".
2024-06-18 12:23:21.839 thread(1) tag(0) DETAIL attr_create_from_str:158 Key-value pair "plc=omron-njnx".
2024-06-18 12:23:21.839 thread(1) tag(0) DETAIL attr_create_from_str:173 Key-value pair before trimming "plc":"omron-njnx".
2024-06-18 12:23:21.839 thread(1) tag(0) DETAIL attr_create_from_str:185 Key-value pair after trimming "plc":"omron-njnx".
2024-06-18 12:23:21.839 thread(1) tag(0) DETAIL attr_create_from_str:158 Key-value pair "elem_size=2".
2024-06-18 12:23:21.839 thread(1) tag(0) DETAIL attr_create_from_str:173 Key-value pair before trimming "elem_size":"2".
[... 305 lines cut ...]
2024-06-18 12:23:21.907 thread(2) tag(0) INFO socket_close:2119 Done.
2024-06-18 12:23:21.907 thread(2) tag(0) INFO socket_destroy:2128 Starting.
2024-06-18 12:23:21.907 thread(2) tag(0) INFO socket_close:2083 Starting.
2024-06-18 12:23:21.907 thread(2) tag(0) INFO socket_close:2119 Done.
2024-06-18 12:23:21.907 thread(2) tag(0) INFO socket_destroy:2145 Done.
[... cut ...]
2024-06-18 12:23:21.909 thread(2) tag(0) DETAIL mutex_destroy:841 destroying mutex 008F9DC0
2024-06-18 12:23:21.909 thread(2) tag(0) DETAIL mutex_destroy:854 Done.
2024-06-18 12:23:21.909 thread(2) tag(0) DETAIL cond_destroy:1242 Starting.
2024-06-18 12:23:21.909 thread(2) tag(0) DETAIL cond_destroy:1253 Done.
2024-06-18 12:23:21.909 thread(2) tag(0) INFO ab_tag_destroy:844 Finished releasing all tag resources.
2024-06-18 12:23:21.909 thread(2) tag(0) INFO ab_tag_destroy:846 done
2024-06-18 12:23:21.911 thread(2) tag(0) INFO refcount_cleanup:268 D
```

MountainKing91, 2024-06-18:
> That's odd. You are not using the mapper at all so nothing should be changing the element count out from underneath you... Can you try that with the tag set to "AlarmList[0]" and capture the debug output?

What size and count? I'm getting lost

timyhac, 2024-06-18:
> Starting with BOOLs seemed like a good idea.
    - https://github.com/libplctag/libplctag.NET/issues/381#issuecomment-2175719103

A DINT array would be more straightforward. BOOL arrays and strings seem to have a lot more variation.

kyle-github, 2024-06-18:
The error above is coming from the PLC.

Sorry, this is getting confusing.  I'll try again.  I thought you were still using the mapper.

tag name: AlarmList[0] <--- the array index of zero is important here
element size = 1
element count = 8

This should get us all eight bytes of the BOOL array.  If the PLC is using backing integers of 16 bits.  If it is not, we should be able to see what it is using.  Note that element size is used only as a hint to pre-size the buffer in the tag memory.

In case it is not clear, when I say "backing integers" I mean that the PLC is storing the bits of a BOOL array in some sort of integer container.  Rockwell decided to use DINTs (32-bit).  According to the manual, Omron uses INTs (16-bit).  This is where the mapper is making an assumption that is incorrect.  It is right for Rockwell, but not Omron.

When I read TestBOOLArray[0], element count = 1 on my Rockwell PLC, I get the following response:

```
d2 00 00 00 d3 00 02 00 00 00
```

d2 00 00 00 - this is the CIP command response header and says that the read completed with no errors.

d3 00  - the base type of the response is d3 which means a 32-bit bit string.  This is what Rockwell uses to store BOOL arrays.

02 00 00 00 - 32 bits of data with the BOOLs packed one to each bit.

What you are seeing from your PLC is this, when we read the whole tag:

```
cc 00 00 00 c1 00 c5 00 00 00 00 00 00 00 
```

cc 00 00 00 - cip command response

c1 00 - the data type is BIT which is only sent as at least a byte (you cannot send just one bit).

c5 00 00 00 00 00 00 00 - 8 bytes of data, probably stored as 16-bit bit strings

The experiment here is to try to use an array index.  That seems to force the PLC into addressing based on individual bits, unlike Rockwell.

A. we get c5 00 00 00 00 00 00 00 again so we get the whole array.

B. we get c5 00 - we get the first 16-bit word, with the only the first 8 bits reported.

C. we get c5.  I don't know what that means other than confusion.  

D. we get 01 00 01 00 00 00 01 01 - we get one byte per bit of the array.

E. The PLC sends us an error.

F. ???

MountainKing91, 2024-06-18:
> In case it is not clear, when I say "backing integers" I mean that the PLC is storing the bits of a BOOL array in some sort of integer container. Rockwell decided to use DINTs (32-bit). According to [...]

Ok, so for Omron:
- a single BOOL tag allocates 16 bit (1 word, 2 bytes), meaning that using single bool tags is "inefficient"
- for BOOL[x] tags (ARRAY 0...x-1 OF BOOL), each BOOL value is stored in a bit of a word, so for example
from BOOL[1] to BOOL[16] --> 2 bytes
from BOOL[17] to BOOL[32] --> 4 bytes
from BOOL[33] to BOOL[64] --> 8 bytes
etc...

is that right @kyle-github? That would make sense when you say that reading my BOOL[64] AlarmList tag we expect 8 bytes of data and we should set ElementCount to 8. I am not sure about ElementSize though.

MountainKing91, 2024-06-18:
Here is a test with an array of DINT, again ErrorBadParam
[image]

ElementSize = 4 <--- each element is a DINT so 4 bytes
ElementCount = 16 <--- there are 16 DINT values in my array
```
const int ARRAY_LENGTH = 16;
const int TIMEOUT = 1000;

var myArrayTag = new Tag()
{
    Name = "DintArray",
    Gateway = "192.168.250.1",
    Path = "1,0",
    PlcType = PlcType.Omron,
    Protocol = Protocol.ab_eip,
    ElementSize = 2,
    ElementCount = ARRAY_LENGTH,
    Timeout = TimeSpan.FromMilliseconds(TIMEOUT),
};

try
{
    myArrayTag.Initialize();
    myArrayTag.Read();
    for (int i = 0; i < ARRAY_LENGTH; i++)
    {
        int arrayDint = myArrayTag.GetInt32(i * 4);
        Console.WriteLine($"Value[{i}]: {arrayDint}");
    }
}
catch (Exception ex)
{
    Console.WriteLine(ex.Message);
}
```
```
2024-06-18 14:18:01.100 thread(1) tag(0) INFO plc_tag_create_ex:838 Starting
2024-06-18 14:18:01.100 thread(1) tag(0) INFO initialize_modules:177 Starting.
2024-06-18 14:18:01.100 thread(1) tag(0) INFO initialize_modules:186 Creating library mutex.
2024-06-18 14:18:01.100 thread(1) tag(0) DETAIL mutex_create:726 Starting.
2024-06-18 14:18:01.100 thread(1) tag(0) DETAIL mutex_create:753 Done.
2024-06-18 14:18:01.100 thread(1) tag(0) INFO initialize_modules:207 Initializing library modules.
2024-06-18 14:18:01.100 thread(1) tag(0) INFO lib_init:136 Starting.
2024-06-18 14:18:01.100 thread(1) tag(0) INFO lib_init:140 Setting up global library data.
2024-06-18 14:18:01.100 thread(1) tag(0) INFO lib_init:142 Creating tag hashtable.
2024-06-18 14:18:01.100 thread(1) tag(0) INFO hashtable_create:75 Starting
2024-06-18 14:18:01.100 thread(1) tag(0) INFO hashtable_create:99 Done
2024-06-18 14:18:01.100 thread(1) tag(0) INFO lib_init:148 Creating tag hashtable mutex.
2024-06-18 14:18:01.100 thread(1) tag(0) DETAIL mutex_create:726 Starting.
2024-06-18 14:18:01.100 thread(1) tag(0) DETAIL mutex_create:753 Done.
2024-06-18 14:18:01.100 thread(1) tag(0) INFO lib_init:154 Creating tag condition variable.
2024-06-18 14:18:01.100 thread(1) tag(0) DETAIL cond_create:1087 Starting.
2024-06-18 14:18:01.100 thread(1) tag(0) DETAIL cond_create:1114 Done.
2024-06-18 14:18:01.100 thread(1) tag(0) INFO lib_init:160 Creating tag tickler thread.
2024-06-18 14:18:01.100 thread(1) tag(0) DETAIL thread_create:884 Starting.
2024-06-18 14:18:01.100 thread(1) tag(0) DETAIL thread_create:917 Done.
2024-06-18 14:18:01.100 thread(1) tag(0) INFO lib_init:166 Done.
2024-06-18 14:18:01.100 thread(1) tag(0) INFO initialize_modules:210 Initializing AB module.
2024-06-18 14:18:01.100 thread(1) tag(0) INFO ab_init:125 Initializing AB protocol library.
2024-06-18 14:18:01.100 thread(1) tag(0) DETAIL mutex_create:726 Starting.
2024-06-18 14:18:01.100 thread(2) tag(0) INFO tag_tickler_func:513 Starting.
2024-06-18 14:18:01.100 thread(1) tag(0) DETAIL mutex_create:753 Done.
2024-06-18 14:18:01.100 thread(1) tag(0) INFO ab_init:134 Finished initializing AB protocol library.
2024-06-18 14:18:01.100 thread(1) tag(0) INFO initialize_modules:215 Initializing Modbus module.
2024-06-18 14:18:01.100 thread(1) tag(0) INFO mb_init:2469 Starting.
2024-06-18 14:18:01.100 thread(1) tag(0) DETAIL mb_init:2471 Setting up mutex.
2024-06-18 14:18:01.100 thread(1) tag(0) DETAIL mutex_create:726 Starting.
2024-06-18 14:18:01.102 thread(1) tag(0) DETAIL mutex_create:753 Done.
2024-06-18 14:18:01.102 thread(1) tag(0) INFO mb_init:2480 Done.
[... 323 lines cut ...]
2024-06-18 14:18:01.176 thread(2) tag(0) INFO socket_close:2119 Done.
2024-06-18 14:18:01.176 thread(2) tag(0) INFO socket_destroy:2128 Starting.
2024-06-18 14:18:01.176 thread(2) tag(0) INFO socket_close:2083 Starting.
2024-06-18 14:18:01.176 thread(2) tag(0) INFO socket_close:2119 Done.
2024-06-18 14:18:01.176 thread(2) tag(0) INFO socket_destroy:2145 Done.
2024-06-18 14:18:01.176 thread(2) tag(0) INFO session_close_socket:804 Done.
2024-06-18 14:18:01.176 thread(2) tag(0) DETAIL session_destroy:885 Destroying session condition variable.
2024-06-18 14:18:01.176 thread(2) tag(0) DETAIL cond_destroy:1242 Starting.
2024-06-18 14:18:01.176 thread(2) tag(0) DETAIL cond_destroy:1253 Done.
2024-06-18 14:18:01.176 thread(2) tag(0) DETAIL session_destroy:892 Destroying session mutex.
2024-06-18 14:18:01.176 thread(2) tag(0) DETAIL mutex_destroy:841 destroying mutex 0085FE58
2024-06-18 14:18:01.176 thread(2) tag(0) DETAIL mutex_destroy:854 Done.
2024-06-18 14:18:01.176 thread(2) tag(0) DETAIL session_destroy:898 Cleaning up allocated memory for paths and host name.
2024-06-18 14:18:01.176 thread(2) tag(0) INFO session_destroy:914 Done.
2024-06-18 14:18:01.178 thread(2) tag(0) INFO refcount_cleanup:268 Done.
2024-06-18 14:18:01.178 thread(2) tag(0) DETAIL mutex_destroy:841 destroying mutex 00864D8C
2024-06-18 14:18:01.178 thread(2) tag(0) DETAIL mutex_destroy:854 Done.
2024-06-18 14:18:01.178 thread(2) tag(0) DETAIL mutex_destroy:841 destroying mutex 00864D90
2024-06-18 14:18:01.178 thread(2) tag(0) DETAIL mutex_destroy:854 Done.
2024-06-18 14:18:01.178 thread(2) tag(0) DETAIL cond_destroy:1242 Starting.
2024-06-18 14:18:01.178 thread(2) tag(0) DETAIL cond_destroy:1253 Done.
2024-06-18 14:18:01.178 thread(2) tag(0) INFO ab_tag_destroy:844 Finished releasing all tag resources.
2024-06-18 14:18:01.178 thread(2) tag(0) INFO ab_tag_destroy:846 done
2024-06-18 14:18:01.178 thread(2) tag(0) INFO refcount_cleanup:268 Done.
```

kyle-github, 2024-06-18:
Remember that Omron will hand back the entire array if you do not have an index part on it.  So "DintArray" is going to pull back the entire array.    So when you try to read 16 elements there is an error because the entire array is not part of another array.

Omron differentiates between the entire object and elements within the object.  But if you do not have an index part, then it defaults to reading the entire object.   

Try "DintArray[0]" or set element count to 1.

kyle-github, 2024-06-18:
Can you try a similar test with the BOOL array again, but not using the mapper, just Tag?

start with index 0 in the array as with the above test.

elem size = 2
elem count = 4

Print out the bytes we get.  I really do not know what we will see if we can get it to work.  Hopefully there will be enough information to help Tim and others work on the .Net wrapper to make it work on Omron too.

kyle-github, 2024-06-18:
This is making me wonder if I shouldn't weak how Omron is handled so that if there isn't an index given, I put [0] in.

MountainKing91, 2024-06-18:
Yes, here it is (also before it was tag and not mapper)
```
var myArrayTag = new Tag()
{
    Name = "AlarmList[0]",
    Gateway = "192.168.250.1",
    Path = "1,0",
    PlcType = PlcType.Omron,
    Protocol = Protocol.ab_eip,
    ElementSize = 2,
    ElementCount = 4,
    Timeout = TimeSpan.FromMilliseconds(1000),
};
```
Does this work? I see I get a 4 bytes buffer, which seems ok for BOOL[64]
```
2024-06-18 16:28:39.002 thread(1) tag(0) INFO plc_tag_create_ex:838 Starting
2024-06-18 16:28:39.004 thread(1) tag(0) INFO initialize_modules:177 Starting.
2024-06-18 16:28:39.004 thread(1) tag(0) INFO initialize_modules:186 Creating library mutex.
2024-06-18 16:28:39.004 thread(1) tag(0) DETAIL mutex_create:726 Starting.
2024-06-18 16:28:39.004 thread(1) tag(0) DETAIL mutex_create:753 Done.
2024-06-18 16:28:39.004 thread(1) tag(0) INFO initialize_modules:207 Initializing library modules.
2024-06-18 16:28:39.004 thread(1) tag(0) INFO lib_init:136 Starting.
2024-06-18 16:28:39.004 thread(1) tag(0) INFO lib_init:140 Setting up global library data.
2024-06-18 16:28:39.004 thread(1) tag(0) INFO lib_init:142 Creating tag hashtable.
2024-06-18 16:28:39.004 thread(1) tag(0) INFO hashtable_create:75 Starting
2024-06-18 16:28:39.004 thread(1) tag(0) INFO hashtable_create:99 Done
2024-06-18 16:28:39.004 thread(1) tag(0) INFO lib_init:148 Creating tag hashtable mutex.
2024-06-18 16:28:39.004 thread(1) tag(0) DETAIL mutex_create:726 Starting.
2024-06-18 16:28:39.004 thread(1) tag(0) DETAIL mutex_create:753 Done.
2024-06-18 16:28:39.004 thread(1) tag(0) INFO lib_init:154 Creating tag condition variable.
2024-06-18 16:28:39.004 thread(1) tag(0) DETAIL cond_create:1087 Starting.
2024-06-18 16:28:39.004 thread(1) tag(0) DETAIL cond_create:1114 Done.
2024-06-18 16:28:39.004 thread(1) tag(0) INFO lib_init:160 Creating tag tickler thread.
2024-06-18 16:28:39.004 thread(1) tag(0) DETAIL thread_create:884 Starting.
2024-06-18 16:28:39.004 thread(1) tag(0) DETAIL thread_create:917 Done.
2024-06-18 16:28:39.004 thread(1) tag(0) INFO lib_init:166 Done.
2024-06-18 16:28:39.004 thread(1) tag(0) INFO initialize_modules:210 Initializing AB module.
2024-06-18 16:28:39.004 thread(1) tag(0) INFO ab_init:125 Initializing AB protocol library.
2024-06-18 16:28:39.004 thread(1) tag(0) DETAIL mutex_create:726 Starting.
2024-06-18 16:28:39.004 thread(2) tag(0) INFO tag_tickler_func:513 Starting.
2024-06-18 16:28:39.004 thread(1) tag(0) DETAIL mutex_create:753 Done.
2024-06-18 16:28:39.004 thread(1) tag(0) INFO ab_init:134 Finished initializing AB protocol library.
2024-06-18 16:28:39.004 thread(1) tag(0) INFO initialize_modules:215 Initializing Modbus module.
2024-06-18 16:28:39.004 thread(1) tag(0) INFO mb_init:2469 Starting.
2024-06-18 16:28:39.004 thread(1) tag(0) DETAIL mb_init:2471 Setting up mutex.
2024-06-18 16:28:39.004 thread(1) tag(0) DETAIL mutex_create:726 Starting.
2024-06-18 16:28:39.004 thread(1) tag(0) DETAIL mutex_create:753 Done.
2024-06-18 16:28:39.004 thread(1) tag(0) INFO mb_init:2480 Done.
2024-06-18 16:28:39.004 thread(1) tag(0) INFO initialize_modules:226 Done initializing library modules.
2024-06-18 16:28:39.004 thread(1) tag(0) INFO initialize_modules:231 Done.
2024-06-18 16:28:39.006 thread(1) tag(0) DETAIL attr_create_from_str:129 Starting.
2024-06-18 16:28:39.006 thread(1) tag(0) DETAIL attr_create_from_str:158 Key-value pair "protocol=ab_eip".
2024-06-18 16:28:39.006 thread(1) tag(0) DETAIL attr_create_from_str:173 Key-value pair before trimming "protocol":"ab_eip".
2024-06-18 16:28:39.006 thread(1) tag(0) DETAIL attr_create_from_str:185 Key-value pair after trimming "protocol":"ab_eip".
2024-06-18 16:28:39.008 thread(1) tag(0) DETAIL attr_create_from_str:158 Key-value pair "gateway=192.168.250.1".
2024-06-18 16:28:39.008 thread(1) tag(0) DETAIL attr_create_from_str:173 Key-value pair before trimming "gateway":"192.168.250.1".
2024-06-18 16:28:39.008 thread(1) tag(0) DETAIL attr_create_from_str:185 Key-value pair after trimming "gateway":"192.168.250.1".
2024-06-18 16:28:39.008 thread(1) tag(0) DETAIL attr_create_from_str:158 Key-value pair "path=1,0".
2024-06-18 16:28:39.008 thread(1) tag(0) DETAIL attr_create_from_str:173 Key-value pair before trimming "path":"1,0".
2024-06-18 16:28:39.008 thread(1) tag(0) DETAIL attr_create_from_str:185 Key-value pair after trimming "path":"1,0".
2024-06-18 16:28:39.008 thread(1) tag(0) DETAIL attr_create_from_str:158 Key-value pair "plc=omron-njnx".
2024-06-18 16:28:39.008 thread(1) tag(0) DETAIL attr_create_from_str:173 Key-value pair before trimming "plc":"omron-njnx".
2024-06-18 16:28:39.008 thread(1) tag(0) DETAIL attr_create_from_str:185 Key-value pair after trimming "plc":"omron-njnx".
2024-06-18 16:28:39.008 thread(1) tag(0) DETAIL attr_create_from_str:158 Key-value pair "elem_size=2".
2024-06-18 16:28:39.008 thread(1) tag(0) DETAIL attr_create_from_str:173 Key-value pair before trimming "elem_size":"2".
2024-06-18 16:28:39.008 thread(1) tag(0) DETAIL attr_create_from_str:185 Key-value pair after trimming "elem_size":"2".
2024-06-18 16:28:39.008 thread(1) tag(0) DETAIL attr_create_from_str:158 Key-value pair "elem_count=4".
2024-06-18 16:28:39.010 thread(1) tag(0) DETAIL attr_create_from_str:173 Key-value pair before trimming "elem_count":"4".
2024-06-18 16:28:39.010 thread(1) tag(0) DETAIL attr_create_from_str:185 Key-value pair after trimming "elem_count":"4".
[... 321 lines cut ...]
2024-06-18 16:28:39.076 thread(3) tag(11) DETAIL unpack_response:1666 Done.
[... cut ...]
2024-06-18 16:28:39.078 thread(1) tag(11) DETAIL tag_raise_event:218 PLCTAG_EVENT_READ_COMPLETED raised with status PLCTAG_STATUS_OK.
2024-06-18 16:28:39.078 thread(1) tag(11) INFO plc_tag_read:1724 elapsed time 14ms
2024-06-18 16:28:39.078 thread(2) tag(11) DETAIL tag_tickler_func:550 Tickling tag 11.
2024-06-18 16:28:39.078 thread(1) tag(0) INFO plc_tag_read:1737 Done
2024-06-18 16:28:39.078 thread(2) tag(11) DETAIL plc_tag_generic_tickler:281 Tickling tag 11.
2024-06-18 16:28:39.080 thread(2) tag(11) DETAIL plc_tag_generic_tickler:386 Done.
00 00 00 00
```

MountainKing91, 2024-06-18:
Nope, I would expect 8 bytes, not 4 =/

kyle-github, 2024-06-18:
Well that's not what I expected.   Even if that is one byte per bit, there are none that are set to 1.  It looks like Omron is exposing only type C1 instead of whatever it uses internally to back the BOOL array.

Is the array all zeros in the PLC?  All BOOLs are false?

I think there are two options to what the PLC is doing.  

A. the PLC is giving us single bytes at a time packed with bits.  In this case, we got the first four bytes of the array (which is 8 bytes long).  In this case, the element count is counting single bytes.

B. the PLC is returning a single byte per BOOL we asked for.  Here we got four bytes that represent the first four bits of the array.  In this case the element count is counting single bits.

What would help us tell the difference?   

If using an element count greater than 8 and less than or equal to 64 works, then it is case B.   Also, if you can make sure that some of the BOOLs are set in the array, that would help.  Then we can see some non-zero data.

Can you try this again (sorry!) with an element count of 10?   If we are counting bytes as elements, then this should fail.  If we are counting bits as elements, then this should succeed.

MountainKing91, 2024-06-18:
> Is the array all zeros in the PLC? All BOOLs are false?

Yes

Here is the buffer when all even elements are set to TRUE, the others are set to FALSE
`01 00 01 00`

kyle-github, 2024-06-18:
Ah, it looks like it is case B.   

Here's what I found in the Omron manual!

[image]

So when we request part of an array of BOOL, we get single bytes back.  One for each bit.  The values will be 0x01 for true an 0x00 for false.

So, your tests above agree with the manual (that is not always the case with these things).

There are two interesting things here.  

1. The format of BOOL arrays is completely different from Rockwell.  Not even slightly similar.
2. When going to write, it looks like we are missing a length field after the data type.  That field is not there with Rockwell.  Another protocol difference.  Strangely this seems to mean that when writing the element count is in the request twice.

kyle-github, 2024-06-18:
Oh, and the fun part: When you read the BOOL array as a full object (no index segment after the name) then you get all the bits packed into bytes.  The format is different when you have an index.

Strings do something similar IIRC.

MountainKing91, 2024-06-18:
All right, asking for ElementCount = 64 gives the correct result, one byte for each bit like you said
`01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00`

MountainKing91, 2024-06-18:
To clear things up

AlarmList is BOOL[64], all even elements are set to TRUE, the others are set to FALSE

**Without index in the tag name**

```
var myArrayTag = new Tag()
{
    Name = "AlarmList",
    Gateway = "192.168.250.1",
    Path = "1,0",
    PlcType = PlcType.Omron,
    Protocol = Protocol.ab_eip,
    ElementSize = 1,
    ElementCount = 1,
    Timeout = TimeSpan.FromMilliseconds(1000),
};

myArrayTag.Initialize();
myArrayTag.Read();
byte[] buffer = myArrayTag.GetBuffer();

string hexString = BitConverter.ToString(buffer);
Console.WriteLine(hexString.Replace('-', ' '));

bool[] rc = new bool[64];
BitArray bitArray = new BitArray(buffer);

for (int i = 0; i < rc.Length; i++)
{
    rc[i] = bitArray.Get(i);
    Console.WriteLine($"Value[{i}]: {rc[i]}");
}
```
Buffer: `55 55 55 55 55 55 55 55`
The array elements are packed into bytes.

**With index in the tag name**

```
var myArrayTag = new Tag()
{
    Name = "AlarmList[0]",
    Gateway = "192.168.250.1",
    Path = "1,0",
    PlcType = PlcType.Omron,
    Protocol = Protocol.ab_eip,
    ElementSize = 1,
    ElementCount = 64,
    Timeout = TimeSpan.FromMilliseconds(1000),
};

myArrayTag.Initialize();
myArrayTag.Read();
byte[] buffer = myArrayTag.GetBuffer();

string hexString = BitConverter.ToString(buffer);
Console.WriteLine(hexString.Replace('-', ' '));

bool[] rc = new bool[64];

for (int i = 0; i < rc.Length; i++)
{
    rc[i] = buffer[i] == 0x1 ? true : false;
    Console.WriteLine($"Value[{i}]: {rc[i]}");
}
```
Buffer: `01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00 01 00`
Each byte represents a bit element of the array, `0x1` means TRUE, `0x0` means FALSE.

Both ways works.

Now, what about the mapper? ^^

kyle-github, 2024-06-18:
It is certainly less data when all the bits are packed into the bytes.  But the fact that you get two different formats is really painful.

kyle-github, 2024-06-18:
Hmm, I wonder what you get when the BOOL array is part of a UDT?

MountainKing91, 2024-06-18:
Let's see;

[image]
[image]

```
var myTag = new Tag()
{
    Name = "BoolArrayInUdt",
    Gateway = "192.168.250.1",
    Path = "1,0",
    PlcType = PlcType.Omron,
    Protocol = Protocol.ab_eip,
    Timeout = TimeSpan.FromMilliseconds(1000),
};
```
Buffer:
`55 55 00 00 FF 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00`

It seems the 16 bits are packed in the first 2 bytes of the buffer, what about the following two bytes before FF?

timyhac, 2024-06-18:
In terms of developing a Mapper for this unfortunately the Mapper does't know the tag name, so you will need to have a different mapper for each format - or just choose one and use that.

If using the one byte per bool format 
* GetElementCout => Product of Array Dimensions
* Decode => value == 0x01
* Encode => value ? 0x01 : 0x00

If using the packed format
* GetElementCount => Product of Array Dimensions / 16
* Decode/Encode => Use `GetBit` and `SetBit` to extract the bit from the byte array

Studying the existing Mappers should give you an idea of how this works:
https://github.com/libplctag/libplctag.NET/blob/master/src/libplctag/DataTypes/BoolPlcMapper.cs

kyle-github, 2024-06-18:
I think i will need to implement another attribute on a tag.  Something like "tag_type_guess".  The internals of the core DLL know the tag name and I can capture the last segment type of it when setting up the tag.   Once we have the data type (C1) comming in from the PLC, I should be able to guess fairly well.  It might take some tweaking.

MountainKing91, 2024-06-18:
@timyhac i'll look at the mappers and see where I can get too. I understand that at this point it's worth testing each and every non trivial variable with a Tag and after that eventually define a custom mapper.

@kyle-github  I noticed something else, probably worth to be pointed out, looking into a UDT with a BOOL value and a STRING[256].
The entire udt takes 400 bytes.
[image]
[image]

With Tag I can explore the buffer
```
var myTag = new Tag()
{
    Name = "SelectedStepCfg",
    Gateway = "192.168.250.1",
    Path = "1,0",
    PlcType = PlcType.Omron,
    Protocol = Protocol.ab_eip,
    Timeout = TimeSpan.FromMilliseconds(1000),
};
```

```
01 01 61 62 63 64 65 66 67 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 00 00 00 C4 09 00 00 00 00 00 00 88 13 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
```

byte 0 --> 01 --> Included = TRUE
byte 1 to 259 --> 61 62 63 64 65 66 67 00 00 00 .. --> Description = abcdefg
byte 260 to 263 --> 04 00 00 00 --> enum Mode = 4, it's four bytes because it is really a DINT
byte 264 to 271 --> C4 09 00 00 00 00 00 00 --> Wait = 2500, a LINT
byte 272 to 279 --> 88 13 00 00 00 00 00 00 --> Timeout = 5000, a LINT
etc..
[image]

It looks like in this situation the BOOL variable takes 1 byte, the actual value of the STRING[256] is offset by 1 byte, which I imagine contains the string length, 256.

Does this make sense?

timyhac, 2024-06-18:
Another example that could be useful is here: https://github.com/libplctag/libplctag.NET/blob/master/src/Examples/CSharp%20DotNetCore/SequencePlcMapper.cs

MountainKing91, 2024-06-18:
@timyhac yes, I have written my mappers looking exactly at that, it was very useful

kyle-github, 2024-06-18:
I found that apparently BOOLs take two bytes when they are single BOOLs.

[image]

See Variable C in the picture.   There is an entirely empty byte there.   That begs the question. What is going on with the first two bytes in the UDT?  They are both 01.  The second one doesn't have anything to do with the string, I think.

According to the alignment rules, those LINTs should be aligned on 8 byte boundaries but the Mode DINT is right up against Wait. Maybe the string ended up having some padding at the end?

Here are the alignment rules:

[image]

I think that there is padding between the string and the Mode.   The string is 256 bytes, possibly 257 depending on how you read the string part of the alignment rules.   There were two bytes for the first BOOL.  So that's 258 or 259.  That's going to align the Mode at a four byte boundary and the next one is at byte 260.  So there is at least one padding byte.   The LINTs have to be aligned at 8-byte boundaries, and the byte after the Mode is byte 264 which is divisible by 8, so it is already aligned.

Bytes
0..1 - BOOL
2-257? 258?  String
257/258..259 Padding
260-263 - Mode DINT
264-271 - Wait LINT
272-279 - Timeout LINT
...

MountainKing91, 2024-06-18:
@kyle-github 
This is the buffer with everything 0 or empty
`00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00`
[image]

This is way I was saying

> It looks like in this situation the BOOL variable takes 1 byte, the actual value of the STRING[256] is offset by 1 byte, which I imagine contains the string length, 256; here I would have expected 2 bytes and not 1...

kyle-github, 2024-06-18:
So the rules for BOOLs (hey that rhymes) are:

1. If the BOOL is a singleton, it takes up 2 bytes.   It will be aligned to 2 bytes.
2. if the BOOL is a subset of a BOOL array in a read response, each bit will take up one byte but the total array size is a multiple of 2 bytes.
3. If the entire array is read or a UDT containing a BOOL array is read, the BOOL is packed as for an array and still aligned at 2 bytes.
4. UDTs are aligned to the alignment of the member with the largest alignment.

MountainKing91, 2024-06-18:
What could be a 'test' UDT to look deeper into this? Multiple BOOL values and a BOOL[xxx]?

MountainKing91, 2024-06-18:
With my UDT
`00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 00 01 00 01 00 01 00 00 00 00 00 00 40 8F 40 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00`

[image]

[image]

kyle-github, 2024-06-18:
It should be if they can be up to about 2k long!   Certainly the SHORT_STRING type (one byte count word) is not going to work.  If that is the case though, then the first two bytes in the result are the count word.   And there's no sign of the BOOL.

With Rockwell, each field has an offset in the structure.   Bit/BOOL fields additionally have a bit number.  The second byte of the type is used for that: C1 04.  The manual did not say anything about that with respect to alignment or memory use for Omron PLCs.

Can you post the debug dump again?  For this case.  I think we're missing something.  All the fields are in order except perhaps the BOOL.  It really looks like that byte right before the string data is not a count byte.

kyle-github, 2024-06-18:
The first byte definitely changed when you toggled the Included field.  The second one seems to be stuck at 01.  It cannot be part of a count.

Ah, the fine manual seems to say that counts are not transmitted with strings.   But the full set of bytes is sent even if the valid data is smaller.

That still leaves us with that pesky extra 01 byte...

MountainKing91, 2024-06-18:
> The first byte definitely changed when you toggled the Included field
Yes!

> Ah, the fine manual seems to say that counts are not transmitted with strings. But the full set of bytes is sent even if the valid data is smaller.
This don't add up to me, strings are even worse than bools to grasp

dump
```
2024-06-19 00:10:28.714 thread(1) tag(0) INFO plc_tag_create_ex:838 Starting
2024-06-19 00:10:28.715 thread(1) tag(0) INFO initialize_modules:177 Starting.
2024-06-19 00:10:28.715 thread(1) tag(0) INFO initialize_modules:186 Creating library mutex.
2024-06-19 00:10:28.715 thread(1) tag(0) DETAIL mutex_create:726 Starting.
2024-06-19 00:10:28.715 thread(1) tag(0) DETAIL mutex_create:753 Done.
2024-06-19 00:10:28.715 thread(1) tag(0) INFO initialize_modules:207 Initializing library modules.
2024-06-19 00:10:28.715 thread(1) tag(0) INFO lib_init:136 Starting.
2024-06-19 00:10:28.715 thread(1) tag(0) INFO lib_init:140 Setting up global library data.
2024-06-19 00:10:28.715 thread(1) tag(0) INFO lib_init:142 Creating tag hashtable.
2024-06-19 00:10:28.715 thread(1) tag(0) INFO hashtable_create:75 Starting
2024-06-19 00:10:28.715 thread(1) tag(0) INFO hashtable_create:99 Done
2024-06-19 00:10:28.715 thread(1) tag(0) INFO lib_init:148 Creating tag hashtable mutex.
2024-06-19 00:10:28.715 thread(1) tag(0) DETAIL mutex_create:726 Starting.
2024-06-19 00:10:28.715 thread(1) tag(0) DETAIL mutex_create:753 Done.
2024-06-19 00:10:28.715 thread(1) tag(0) INFO lib_init:154 Creating tag condition variable.
2024-06-19 00:10:28.715 thread(1) tag(0) DETAIL cond_create:1087 Starting.
2024-06-19 00:10:28.715 thread(1) tag(0) DETAIL cond_create:1114 Done.
2024-06-19 00:10:28.715 thread(1) tag(0) INFO lib_init:160 Creating tag tickler thread.
2024-06-19 00:10:28.715 thread(1) tag(0) DETAIL thread_create:884 Starting.
2024-06-19 00:10:28.715 thread(1) tag(0) DETAIL thread_create:917 Done.
2024-06-19 00:10:28.715 thread(1) tag(0) INFO lib_init:166 Done.
2024-06-19 00:10:28.715 thread(1) tag(0) INFO initialize_modules:210 Initializing AB module.
2024-06-19 00:10:28.715 thread(1) tag(0) INFO ab_init:125 Initializing AB protocol library.
2024-06-19 00:10:28.715 thread(1) tag(0) DETAIL mutex_create:726 Starting.
2024-06-19 00:10:28.715 thread(2) tag(0) INFO tag_tickler_func:513 Starting.
2024-06-19 00:10:28.715 thread(1) tag(0) DETAIL mutex_create:753 Done.
2024-06-19 00:10:28.715 thread(1) tag(0) INFO ab_init:134 Finished initializing AB protocol library.
2024-06-19 00:10:28.715 thread(1) tag(0) INFO initialize_modules:215 Initializing Modbus module.
2024-06-19 00:10:28.715 thread(1) tag(0) INFO mb_init:2469 Starting.
2024-06-19 00:10:28.715 thread(1) tag(0) DETAIL mb_init:2471 Setting up mutex.
2024-06-19 00:10:28.715 thread(1) tag(0) DETAIL mutex_create:726 Starting.
2024-06-19 00:10:28.716 thread(1) tag(0) DETAIL mutex_create:753 Done.
2024-06-19 00:10:28.716 thread(1) tag(0) INFO mb_init:2480 Done.
2024-06-19 00:10:28.716 thread(1) tag(0) INFO initialize_modules:226 Done initializing library modules.
2024-06-19 00:10:28.716 thread(1) tag(0) INFO initialize_modules:231 Done.
2024-06-19 00:10:28.716 thread(1) tag(0) DETAIL attr_create_from_str:129 Starting.
2024-06-19 00:10:28.716 thread(1) tag(0) DETAIL attr_create_from_str:158 Key-value pair "protocol=ab_eip".
2024-06-19 00:10:28.716 thread(1) tag(0) DETAIL attr_create_from_str:173 Key-value pair before trimming "protocol":"ab_eip".
2024-06-19 00:10:28.717 thread(1) tag(0) DETAIL attr_create_from_str:185 Key-value pair after trimming "protocol":"ab_eip".
2024-06-19 00:10:28.717 thread(1) tag(0) DETAIL attr_create_from_str:158 Key-value pair "gateway=192.168.250.1".
2024-06-19 00:10:28.717 thread(1) tag(0) DETAIL attr_create_from_str:173 Key-value pair before trimming "gateway":"192.168.250.1".
2024-06-19 00:10:28.717 thread(1) tag(0) DETAIL attr_create_from_str:185 Key-value pair after trimming "gateway":"192.168.250.1".
2024-06-19 00:10:28.717 thread(1) tag(0) DETAIL attr_create_from_str:158 Key-value pair "path=1,0".
2024-06-19 00:10:28.717 thread(1) tag(0) DETAIL attr_create_from_str:173 Key-value pair before trimming "path":"1,0".
2024-06-19 00:10:28.717 thread(1) tag(0) DETAIL attr_create_from_str:185 Key-value pair after trimming "path":"1,0".
2024-06-19 00:10:28.718 thread(1) tag(0) DETAIL attr_create_from_str:158 Key-value pair "plc=omron-njnx".
2024-06-19 00:10:28.718 thread(1) tag(0) DETAIL attr_create_from_str:173 Key-value pair before trimming "plc":"omron-njnx".
2024-06-19 00:10:28.718 thread(1) tag(0) DETAIL attr_create_from_str:185 Key-value pair after trimming "plc":"omron-njnx".
2024-06-19 00:10:28.718 thread(1) tag(0) DETAIL attr_create_from_str:158 Key-value pair "name=SelectedStepCfg".
2024-06-19 00:10:28.718 thread(1) tag(0) DETAIL attr_create_from_str:173 Key-value pair before trimming "name":"SelectedStepCfg".
2024-06-19 00:10:28.718 thread(1) tag(0) DETAIL attr_create_from_str:185 Key-value pair after trimming "name":"SelectedStepCfg".
2024-06-19 00:10:28.718 thread(1) tag(0) DETAIL attr_create_from_str:216 Done.
2024-06-19 00:10:28.719 thread(1) tag(0) INFO find_tag_create_func:99 Matched protocol=ab_eip
2024-06-19 00:10:28.719 thread(1) tag(0) INFO ab_tag_create:175 Starting.
2024-06-19 00:10:28.719 thread(1) tag(0) INFO rc_alloc_impl:111 Starting, called from ab_tag_create:182
2024-06-19 00:10:28.719 thread(1) tag(0) INFO rc_alloc_impl:130 Done
2024-06-19 00:10:28.719 thread(1) tag(0) DETAIL rc_alloc_impl:135 Returning memory pointer 00EFEE00
[... cut ...] 0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 00 00 00 C4 09 00 00 00 00 00 00 88 13 00 00 00 00 00 00 01 00 01 00 01 00 01 00 00 00 00 00 00 40 8F 40 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 58 9B 40 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
```

kyle-github, 2024-06-18:
I wonder if we are chasing a red herring.  What if that byte is 1 because there is garbage in the data in memory?  It is effectively padding.

MountainKing91, 2024-06-18:
[image]

The non-evidentiated bytes are for alignment, correct?

MountainKing91, 2024-06-18:
> I wonder if we are chasing a red herring. What if that byte is 1 because there is garbage in the data in memory? It is effectively padding.

How can I make sure of this? I tried power cycling the plc, it doesn't change anything.

kyle-github, 2024-06-18:
I think you were close.   

[image]

So there can be a count between the data type and the data. But it is the number of elements so can only be one here.

But it should be 01 00 not 00 01.  

I am baffled.

kyle-github, 2024-06-18:
Back to your question about alignment.  I think there should be some alignment padding bytes at the end of the string data.  I think it is the last two bytes in the area you marked as part of the string.

MountainKing91, 2024-06-19:
So, in this image I have marked with different colors all the different areas in the buffer

- The first two bytes I don't understand, the first definitely changes when toggling "Included", but the second is always 01 no matter what. We would expect them to be fully dedicated to Included, 00 00 if FALSE and 01 00 if TRUE
- The next 258 bytes seem dedicated to the STRING[256] "Description", but why 258? Padding? When I write something in Description, the 1st character goes into the 3rd byte (the 1st of the dark green range)... example: "abcdefg" --> from the 3rd byte going forward I have 61 62 63 64 65 66 67 00 00 00 ...
- The BOOL values in the middle (ligth and dark grey) they all occupies 2 bytes each, as expected at this point for BOOL inside a UDT
- Everything else looks fine, apart from bytes 340-343 and bytes 396-399, which are "blank" but I assume it is the alignement of the UDT AxisConfiguration, following the rule specified in the w506 manual

[image]

[image]

[25 more replies]

closed 2024-06-24 by timyhac