libplctag/libplctag.NET #382: Using `raw_tag_type_bytes` for dynamic tags

timyhac, 2024-06-17:
With the release of [libplctag v2.6.0](https://github.com/libplctag/libplctag/releases/tag/v2.6.0), a new use case has been enabled that allows application code to make use of tag metadata sent by the PLC.

This means that it is possible to have dynamically-typed tags without needing to fetch the entire PLC tag listing, which might be useful for some applications where the user is interactively deciding which tags to monitor.
It is also useful if migrating from another PLC communication package which had this capability.

Below is a prototype of what using this capability could look like.

Notes:
* It hasn't been tested - I don't have access to any devices.
* It only supports the simplest of atomic types. Arrays/UDTs/other atomic types are not supported.
* This only works if the PLC returns the type metadata, and the structure of that metadata is the same.
* Error handling - if user code sets Value with an inappropriate type (either before initialization or after), when should the error be thrown - as early as possible or only when communicating with PLC? This could happen for example if the PLC type is `INT` (16bits) but the C# type is `int` (32bits). Should the value be downcast? Do different applications have different answers to this question?

```csharp
public class TagDynamic
{
    private readonly Tag _tag;
    private Func<Tag, int, object> decoder;
    private Action<Tag, int, object> encoder;

    public TagDynamic()
        : base()
    {
        _tag = new Tag()
        {
            PlcType = libplctag.PlcType.ControlLogix,   // Only certain types of devices support this, so hard-code this
            Protocol = Protocol.ab_eip,    // Only certain types of devices support this, so hard-code this
        };

        _tag.ReadCompleted += (s, e) => ReadCompleted?.Invoke(this, e);
        _tag.WriteCompleted += (s, e) => WriteCompleted?.Invoke(this, e);
    }

    /// <inheritdoc cref="Tag.Initialize">
    public void Initialize()
    {
        _tag.Initialize();

        // After we have read the tag metadata, configure the encoder/decoder
        Configure();
    }

    /// <inheritdoc cref="Tag.Write">
    public void Write()
    {
        if (!_tag.IsInitialized)
        {
            Initialize();
        }

       _tag.Write();
    }

    /// <inheritdoc cref="Tag.Read">
    public void Read()
    {
        if (!_tag.IsInitialized)
        {
            Initialize();
            return; // Initialize already does a read, so in this case we don't need to read again and can return early.
        }

        _tag.Read();
    }

    /// <summary>
    /// The value stored in local memory.
    /// </summary> 
    public object Value
    {
        get => decoder(_tag, 0);
        set => encoder(_tag, 0, value);
    }

    /// <inheritdoc cref="Tag.Gateway"/>
    public string Gateway
    {
        get => _tag.Gateway;
        set => _tag.Gateway = value;
    }

    /// <inheritdoc cref="Tag.Path"/>
    public string Path
    {
        get => _tag.Path;
        set => _tag.Path = value;
    }

    /// <inheritdoc cref="Tag.Name"/>
    public string Name
    {
        get => _tag.Name;
        set => _tag.Name = value;
    }

    /// <inheritdoc cref="Tag.Timeout"/>
    public TimeSpan Timeout
    {
        get => _tag.Timeout;
        set => _tag.Timeout = value;
    }

    /// <inheritdoc cref="Tag.AutoSyncReadInterval"/>
    public TimeSpan? AutoSyncReadInterval
    {
        get => _tag.AutoSyncReadInterval;
        set => _tag.AutoSyncReadInterval = value;
    }

    /// <inheritdoc cref="Tag.AutoSyncWriteInterval"/>
    public TimeSpan? AutoSyncWriteInterval
    {
        get => _tag.AutoSyncWriteInterval;
        set => _tag.AutoSyncWriteInterval = value;
    }

    /// <inheritdoc cref="Tag.DebugLevel"/>
    public DebugLevel DebugLevel
    {
        get => _tag.DebugLevel;
        set => _tag.DebugLevel = value;
    }

    public event EventHandler<TagEventArgs> ReadCompleted;
    public event EventHandler<TagEventArgs> WriteCompleted;

    private void Configure()
    {
        var tagTypeBytes = _tag.GetByteArrayAttribute("raw_tag_type_bytes");
        var tagTypeCode = BitConverter.ToUInt16(tagTypeBytes, 0);   // Endianess?

        switch (tagTypeCode)
        {
            case 0xC1: // BOOL: Boolean value
                decoder = (tag, offset) => tag.GetBit(offset);
                encoder = (tag, offset, value) => tag.SetBit(offset, value);
                break;

            case 0xC2: // SINT: Signed 8-bit integer value
                decoder = (tag, offset) => tag.GetInt8(offset);
                encoder = (tag, offset, value) => tag.SetInt8(offset, value);
                break;

            case 0xC3: // INT: Signed 16-bit integer value
                decoder = (tag, offset) => tag.GetInt16(offset);
                encoder = (tag, offset, value) => tag.SetInt16(offset, value);
                break;

            case 0xC4: // DINT: Signed 32-bit integer value
                decoder = (tag, offset) => tag.GetInt32(offset);
                encoder = (tag, offset, value) => tag.SetInt32(offset, value);
                break;

            case 0xC5: // LINT: Signed 64-bit integer value
                decoder = (tag, offset) => tag.GetInt64(offset);
                encoder = (tag, offset, value) => tag.SetInt64(offset, value);
                break;

            case 0xC6: // USINT: Unsigned 8-bit integer value
                decoder = (tag, offset) => tag.GetUInt8(offset);
                encoder = (tag, offset, value) => tag.SetUInt8(offset, value);
                break;

            case 0xC7: // UINT: Unsigned 16-bit integer value
                decoder = (tag, offset) => tag.GetUInt16(offset);
                encoder = (tag, offset, value) => tag.SetUInt16(offset, value);
                break;

            case 0xC8: // UDINT: Unsigned 32-bit integer value
                decoder = (tag, offset) => tag.GetUInt32(offset);
                encoder = (tag, offset, value) => tag.SetUInt32(offset, value);
                break;

            case 0xC9: // ULINT: Unsigned 64-bit integer value
                decoder = (tag, offset) => tag.GetUInt64(offset);
                encoder = (tag, offset, value) => tag.SetUInt64(offset, value);
                break;

            case 0xCA: // REAL: 32-bit floating point value, IEEE format
                decoder = (tag, offset) => tag.GetFloat32(offset);
                encoder = (tag, offset, value) => tag.SetFloat32(offset, value);
                break;

            case 0xCB: // LREAL: 64-bit floating point value, IEEE format
                decoder = (tag, offset) => tag.GetFloat64(offset);
                encoder = (tag, offset, value) => tag.SetFloat64(offset, value);
                break;

            case 0xCC: // Synchronous time value
            case 0xCD: // Date value
            case 0xCE: // Time of day value
            case 0xCF: // Date and time of day value
            case 0xD0: // Character string, 1 byte per character
            case 0xD1: // 8-bit bit string
            case 0xD2: // 16-bit bit string
            case 0xD3: // 32-bit bit string
            case 0xD4: // 64-bit bit string
            case 0xD5: // Wide char character string, 2 bytes per character
            case 0xD6: // High resolution duration value
            case 0xD7: // Medium resolution duration value
            case 0xD8: // Low resolution duration value
            case 0xD9: // N-byte per char character string
            case 0xDA: // Counted character sting with 1 byte per character and 1 byte length indicator
            case 0xDB: // Duration in milliseconds
            case 0xDC: // CIP path segment(s)
            case 0xDD: // Engineering units
            case 0xDE: // International character string (encoding?)
                throw new NotImplementedException();
        }
    }

}
```

And it would be used similar to how strongly typed tags are used.
```csharp
var tag = new DynamicTag()
{
    Name = "MyDintTag",
    // Configure other properties
};
tag.Initialize(); // Detects the tag is a DINT and configures the appropriate getter and setter.
Console.WriteLine(tag.Value);  // Prints e.g. 1234
tag.Value = 4321;
tag.Write();
```

INondas, 2024-06-24:
Hello gurus, the release is applied also in nuget package for c#?  I would like to test it on ControlNet using backplane routing.

timyhac, 2024-06-24:
@INondas - no its just a code example for now.

You can test it by copying the code into your own project.