libplctag/libplctag.NET #384: Expose `max_requests_in_flight` on `Tag` or `Tag<M,T>`.

timyhac, 2024-06-21:
`max_requests_in_flight` is currently not exposed to `Tag` or `Tag<M,T>`.

This issue requests that is is exposed for those classes.

It is unclear whether this should be able to configured after the tag has been Initialized.

kyle-github, 2024-06-21:
It would be difficult to make this work when setting it lower at runtime.  It could orphan some requests.   Setting it higher could possibly work but is probably not a good idea.  I think it is best to make this one only work at creation time.

kyle-github, 2024-06-21:
BTW, this is exposed in the wiki section on type-specific attributes.  This is a bit of a mess, isn't it?

timyhac, 2024-06-22:
> It would be difficult to make this work when setting it lower at runtime. It could orphan some requests. Setting it higher could possibly work but is probably not a good idea. I think it is best to make this one only work at creation time.

Yes no problem I can check for this. I'm more thinking that it might make more sense to let the core library decide what can be changed after creation time and just pass through the error - rather doing that logic in the wrapper.

> BTW, this is exposed in the wiki section on type-specific attributes.

This is where I saw it - I imagine that this has been there for some time, not sure how I missed it.

timyhac, 2024-07-12:
This has been released in [libplctag v1.5.0](https://github.com/libplctag/libplctag.NET/releases/tag/libplctag-v1.5.0)

closed 2024-07-12 by timyhac