libplctag/libplctag.NET #390: STRUCT / UDT on Omron Device

timyhac, 2024-07-08:
@MountainKing91 
Using the below code: **ErrorRemoteErr**

`
var myTag = new Tag()
{
    Name = "BACK_PRESSURE_RELIEF_VALVE_ANIMATION_SEQUENCE_REGISTER",
    Gateway = [my IP],
    Path = "1,0",
    PlcType = PlcType.Omron,
    Protocol = Protocol.ab_eip,
    Timeout = TimeSpan.FromMilliseconds(2000),
};
`
[read array Jul_07_2024_224933.txt](https://github.com/user-attachments/files/16122183/read.array.Jul_07_2024_224933.txt)

The attached video is the real-time values of this tag in keepware, Not sure if it helps or not:

https://github.com/libplctag/libplctag/assets/32189853/e493b5f6-fe79-4d3c-8734-fdaf90b8e1f6

Please let me know if you need any logs.
Thank you very much

_Originally posted by @hamduc7 in https://github.com/libplctag/libplctag/issues/466#issuecomment-2212880475_

timyhac, 2024-07-08:
@hamduc7 - could it just be the letter case on the Tag name?

hamduc7, 2024-07-08:
Hi @timyhac 
I tried **Back_Pressure_Relief_Valve_Animation_Sequence_Register**, error **ErrorRemoteErr**

`
var myTag = new Tag()
                {
                    Name = "Back_Pressure_Relief_Valve_Animation_Sequence_Register",
                    Gateway = IP,
                    Path = path,
                    PlcType = PlcType.Omron,
                    Protocol = Protocol.ab_eip,
                    DebugLevel = DebugLevel.Detail, 
                    Timeout = TimeSpan.FromMilliseconds(1000),
                };
`
[read array Jul_07_2024_235318.txt](https://github.com/user-attachments/files/16122468/read.array.Jul_07_2024_235318.txt)

So looks like it is not about case sensitive.

hamduc7, 2024-07-08:
Dear All,
I found interesting information about bool array in the document!

[data types in omron nj.pdf](https://github.com/user-attachments/files/16122520/data.types.in.omron.nj.pdf)

hamduc7, 2024-07-08:
Use **Back_Pressure_Relief_Valve_Animation_Sequence_Register[0],** error **ErrorBadParam**
`
var myTag = new Tag()
                {
                    Name = "Back_Pressure_Relief_Valve_Animation_Sequence_Register[0]",
                    Gateway = IP,
                    Path = path,
                    PlcType = PlcType.Omron,
                    Protocol = Protocol.ab_eip,
                    DebugLevel = DebugLevel.Detail, 
                    Timeout = TimeSpan.FromMilliseconds(1000),
                };
`
[read array Jul_08_2024_000607.txt](https://github.com/user-attachments/files/16122539/read.array.Jul_08_2024_000607.txt)

timyhac, 2024-07-08:
@hamduc7 - do you have an example of accessing a Tag that does work?

hamduc7, 2024-07-08:
Yes, I can read a single tag. In this example, I read a tag, value = 198.5625. It is same value as keepware

`
var myTag = new libplctag.Tag<libplctag.DataTypes.RealPlcMapper, float>()
                {
                    Name = @"Deaerator_Transfer_Pump_Pressure_PV_Scaled_REAL",
                    Gateway = IP,
                    Path = path,
                    PlcType = libplctag.PlcType.Omron,
                    Protocol = libplctag.Protocol.ab_eip,
                    DebugLevel = libplctag.DebugLevel.None,
                    Timeout = TimeSpan.FromMilliseconds(1000),
                };
`
[read float Jul_08_2024_001856.txt](https://github.com/user-attachments/files/16122633/read.float.Jul_08_2024_001856.txt)

timyhac, 2024-07-08:
Ah so this is actually a Union type, not a Bool Array or a struct.

```
Tag Back_Pressure_Relief_Valve_Animation_Sequence_Register (864):
	Sending request data:  01 1C 91 36 42 61 63 6B 5F 50 72 65 73 73 75 72 65 5F 52 65 6C 69 65 66 5F 56 61 6C 76 65 5F 41 6E 69 6D 61 74 69 6F 6E 5F 53 65 71 75 65 6E 63 65 5F 52 65 67 69 73 74 65 72
	Received response data:  81 00 00 00 08 00 00 00 0C 00 00 00 62 00 00 00 00 00 00 00 00 04 00 00 62 00 00 00
	Tag type 12
```

Tag Type 12 = 0C, which according to the Omron manual is a Union type, not a Struct

[image]

I don't know how these work, I wonder if you have to use the member name in order to access the data i.e. if you want to access it as "Bit_data" you have to specify that in the name.

hamduc7, 2024-07-08:
@timyhac So tag name = Back_Pressure_Relief_Valve_Animation_Sequence_Register.Bit_Data
or tag name = Back_Pressure_Relief_Valve_Animation_Sequence_Register.Bit_Data[0]?

hamduc7, 2024-07-08:
@timyhac Oh, looks likes it works?
`
 var myTag = new Tag()
                {
                    Name = "Back_Pressure_Relief_Valve_Animation_Sequence_Register.Bit_Data",
                    Gateway = IP,
                    Path = path,
                    PlcType = PlcType.Omron,
                    Protocol = Protocol.ab_eip, 
                    Timeout = TimeSpan.FromMilliseconds(1000),
                };
[read array Jul_08_2024_003208.txt](https://github.com/user-attachments/files/16122712/read.array.Jul_08_2024_003208.txt)
`

MountainKing91, 2024-07-08:
I suspected it was a union!
If you can send me its definition in sysmac studio i can reproduce that on my NX controller.

hamduc7, 2024-07-08:
I cannot access sysmac studio.

I tried below code using the new OmronNonIndexedBoolArrayMapper, it works. I cannot compare the data with keepware because it is continuous changed.
`
                var myTag = new Tag<OmronNonIndexedBoolArrayMapper, bool[]>()
                {
                    Name = "BACK_PRESSURE_RELIEF_VALVE_ANIMATION_SEQUENCE_REGISTER.Bit_Data",
                    Gateway = IP,
                    Path = path,
                    PlcType = PlcType.Omron,
                    Protocol = Protocol.ab_eip,
                    Timeout = TimeSpan.FromSeconds(1000),
                };
`

[read array Jul_08_2024_003912.txt](https://github.com/user-attachments/files/16122766/read.array.Jul_08_2024_003912.txt)

timyhac, 2024-07-08:
Great!

MountainKing91, 2024-07-08:
Are you sure the book arrays is 63 bits? If the tag is a union it should be definitely 64 bits.
I will check this on sysmac and let you guys know.

hamduc7, 2024-07-08:
It is an array of 64 elements: 0..63

[image]

MountainKing91, 2024-07-08:
All right, I'll try

timyhac, 2024-07-16:
I trust that this has been resolved!

closed 2024-07-16 by timyhac