libplctag/libplctag.NET #406: Remove Mapper system (`Tag<M,T>`, `IPlcMapper`, `DintPlcMapper`, `TagReal` etc).

timyhac, 2024-07-17:
libplctag handles the details of communicating with PLCs over Ethernet/IP Explicit Messaging, so you can easily access the value of PLC tags.

This statement is mostly true; while it is correct that you (most of the time) do not need to understand Ethernet/IP itself, you do need to understand the device's specific Ethernet/IP API as not all devices are the same.

This is similar to using a HTTP REST API - while you do not need to understand the details of TCP/IP or to some extent even HTTP, you *do* need to know what data to send and how to interpret the response.

Specifically;
* The tag "name"
* How the tag buffer encodes values
* Path/Gateway

A few years after the release of the official libplctag.NET wrapper, a pattern has emerged around issues that library consumers face.
I will be making the argument that the class `Tag<M,T>` (as well as related resources) provided by libplctag.NET has contributed to this issue, and therefore should be removed.

## The good: it handles a common case well!

In `Tag<M,T>`, the `M` is a `PlcMapper` which converts between a byte array (the tag buffer) and the `T` .NET type and makes it possible to implement the Decode/Encode logic for a data type once, and then re-use it for 1D, 2D and 3D arrays as well as the atomic type.
For this use-case it excels.
This system can even be used for UDTs with static sizes.
The value is strongly-typed, which holds significant value in the .NET ecosystem and particularly C#.

It is entirely possible that this serves the silent majority of users perfectly - they have no complaints, and we do not hear from them.
They have used the system as it was intended and moved on with their life.

It's utility stems from being able to instantiate a strongly-typed Tag, simply by providing a Mapper class type (and most of the time this is built-in).
And even this can be hidden because [we also ship "simple" classes](https://github.com/libplctag/libplctag.NET/blob/master/src/libplctag/DataTypes/Simple/Definitions.cs).

## The complexity cliff

`Tag<M,T>` makes a (somewhat implicit) promise that you as a consumer do not need to understand how PLC data is encoded in the buffer, as the `PlcMapper` handles that for you.
However, over time it has become clear that there are additional capabilities that PLCs offer, and that library consumers need, that the Mapper system doesn't fulfil.

At the time of development, the Mapper system was consistent with the core libplctag API, and was universally applicable (at least within the confines of libplctag.NET developers' limited exposure to other PLC vendors).
The key insight was that single values (whether they be atomic or UDTs) and arrays of values can use the same decoding logic.
The logical conclusion is that much of the setup of tags can be contained within `PlcMapper` classes, and `Tag<M,T>` merely needs to expose the Read/Write commands and events.
This abstraction falls off a "cliff" as soon as you go outside the simplest of cases.
But the promise remains - the PlcMapper is *meant* to do it, and it does not - therefore it is a bug in the `PlcMapper` or `Tag<M,T>`.

Ultimately, using libplctag does not preclude you (the library consumer) from understanding the device's Ethernet/IP API.

Consumers will encounter this working
* When working with some of the more complicated types (BOOL Arrays, Strings, UDTs)
* Using some advanced CIP services such as Tag Listing.
* Arrays of arrays (a.k.a. Jagged arrays)
* Dynamic Tags
* When using multiple make/models of device (Allen-Bradley encode some tags different to Omron)

If we want to continue to make the promise that consumers do not need to understand their device's Ethernet/IP API, the only path forward for this project is to wait for complaints to arrive, and incrementally add handlers for specific cases into the mappers.
The mappers would effectively become a documentation store for this knowledge.

## Its another API to learn, support, test and maintain

The good news is that consumers are not locked into using the built-in mappers.
If they are able to understand how their devices behave, they can add custom mappers.

However, this is another API that consumers must learn.
The Mapper system uses some somewhat advanced C# language features which can be tricky for early-career developers to understand and implement.

If there are features that consumers want (such as value change detection), consumers could spend a considerable amount of time trying to determine how libplctag.NET supports it (which it may not), rather than just implementing it themself.

Furthermore:
* libplctag wiki does not apply to most of it.
* There are three wrapper classes plus the mapper - objects which must be garbage collected.
* There are many cases where the returned PLC memory does not have a static size.
* It is code that the libplctag community must document, support and maintain.

It would be simpler if that layer was removed, and consumers directly interfaced with `Tag` - it confronts them with the complexity of the underlying API and does not hide any of the "foot-guns" (which exist whether they can see them or not).

## Many consumers still add their own wrappers

A study of applications or libraries that are consuming libplctag.NET shows that most will create yet another wrapper around `Tag<M,T>`.
This implies that it is not the optimal level of abstraction and doesn't add much value.

## What's the alternative?

Use an abstract base class and handle the specific data types with derived classes.

This has several advantages over the `Tag<M,T>`+`PlcMapper` system:
* Several use-cases would have been immediately enabled (e.g. hooks to add before ad after Initialize is called).
* Consumers can add their own methods such as custom setters/getters that trigger events.

The issue with this is that there is not an obvious universal interface that would serve as the base class.
As an example, should the Tag value be exposed as a .NET type such as `int` or should it be `object` - or should it be exposed at all?
What if the type was an array, and instead of exposing the entire array, the library consumer would prefer to force their application code to go through a method with an [indexer](https://learn.microsoft.com/en-us/dotnet/csharp/programming-guide/indexers/) (so that it can raise events or run other logic).

## Summary

While `Tag<M,T>` has utility for a common use-case, the API it exposes is not the lowest-common denominator.
Having it is a burden on library consumers and maintainers alike.

I recommend:
* Removing [`ITag`](https://github.com/libplctag/libplctag.NET/blob/master/src/libplctag/ITag.cs), [`Tag<M,T>`](https://github.com/libplctag/libplctag.NET/blob/master/src/libplctag/TagOfT.cs), all [Mappers and Simple Tag classes](https://github.com/libplctag/libplctag.NET/tree/master/src/libplctag/DataTypes), and also not supplying an alternative like an abstract base type.
* Instead, moving `Tag<M,T>` and a demonstration of an alternative like an abstract base mapper type to the Examples folder.
* Move [Tag Listing](https://github.com/libplctag/libplctag.NET/blob/master/src/libplctag/DataTypes/TagInfoPlcMapper.cs) for Rockwell PLC to the Examples folder.
* Creating a separate repository within the libplctag organization for known tag buffer encodings for combinations of PlcType, Tag Name, Protocol, etc
* Clarify the promise that libplctag makes to customers - what *should* libplctag be capable of, and what is left up to the consumer to discover?

If this happens, the libplctag.NET would be closer to a "wrapper" in spirit, rather than what could be argued to be an opinionated framework for working with libplctag.

I would be happy to offer a separate Nuget package for the Mapper system, but not as an "official" package by the libplctag organization, instead released under a "timyhac" brand such as `timyhac.libplctag.Mappers`

I've created this as a Github issue to open it up for community feedback. I realize that there are many users of the Mapper system at this point and I want to take into account the impact on them.

trinnan42, 2024-07-17:
I have some concerns with this. (Note: Please correct me if I misunderstood anything in your explanation.)

### UDTs

We're currently using the custom mappers to handle UDTs quite extensively.

How do you imagine this would impact how UDTs are handled?

Would something in this new version of libplctag.NET handle UDTs itself in a way that wouldn't necessitate the mappers? If not, we would need to use this potential `timyhac.libplctag.Mappers` package or maintain these classes ourselves? If i understand correctly `Tag<M,T>` and `IPlcMapper`, would move to be examples?

I'm concerned with needing more dependencies if the latter were the case. I would prefer that it remains part of the main library to needing another library to maintain.

Just to give a little more information regarding your mention of wrappers, we do have our own wrapper around libplctag.NET, but for the purpose of being able to support other PLC manufacturers and protocols like Siemens and OPC UA.

### Wrapper

We have an `IPlcComm` interface and then an implementation for each PLC type, currently `LibPlcTagComm` and `SiemensOpcUaComm`.

We define our tags in an XML file with a generic name that we use throughout our business logic. This name is defined with associated in the XML file, for each PLC driver that we support, with information used to generate the tag with (address, data type, etc.). Then I do some reflection to generate a dynamic typed `Tag<M,T>` instance using the appropriate native data type and its corresponding PlcMapper. It may be a bit messy but it's working quite well for us. (Unfortunately I can't directly provide the code itself...)

Here again, we'd either need to rewrite all of this logic to use just what libplctag.NET provides or use this additional Mapper library and again, I'd prefer if this all just remained part of the library.

timyhac, 2024-07-18:
Hi @trinnan42 - thanks for the feedback!

You are correct in saying you would need to find an alternate implementation for the mapper system - either one you maintain yourself or a separate package. The proposal can be seen in [this branch](https://github.com/libplctag/libplctag.NET/blob/%23406/examples/CSharp%20DotNetCore/PlcMapper/TagOfT.cs).

As I see it, the value of `Tag<M,T>` and the Mapper system is:
1. The tag value is exposed as a C# type rather than a collection of Data Accessors.
2. The mapper logic is a store of knowledge for how the tag bytes are structured, and which Data Accessors to use, for various scenarios.

For point 1, it sounds like you're already doing some logic to convert from a typed value back into `dynamic` - so I don't think you're receiving much of that benefit.

For point 2, frankly I'm doing a terrible job at supporting scenarios other than the fairly specific Rockwell/Atomic tag scenario - and I don't really have a way to improve this other than wait for complaints to roll in.

Do you mind if I ask what value you get out of the `Tag<M,T>` class?

Thanks so much for the detailed response to the post!

timyhac, 2024-07-18:
> Any kind of structured data should not be the core's concern. Strings raise the difficulty even higher because they are inconsistent even across the same hardware. Strings should not be supported in the core or the core wrapper.

@kyle-github - yep 100%, and I think that the libplctag.NET wrapper using the libplctag name makes some implicit promise about what it does, which can be tricky to get to the bottom of when things go wrong.

trinnan42, 2024-07-18:
@timyhac 

Well, to try to explain with more detail what I'm doing without being able to share actual code:

1. I have an Enum which defines the available Data Types.
2. I have some `Dictionary ` instances which map these data types to C# native data types and the Mappers.
3. I create a dynamic tag variable by creating an instance of `Tag<,>` with the types from these dictionaries based on the Data Type Enum's value.
4. I have `Read<T>` and `Write<T>` methods exposed to my code defined with my IPlcComm interface.
5. The Read<T> method instantiates this dynamic instance of `Tag<M,T>` and then I just directly return `Tag<M,T>.Value` for Read and set `Tag<M,T>.Value` for Write.

I guess I'm not exactly sure myself what value I'm getting out of `Tag<M,T>`. When I began using this a couple years ago. I based what I'm doing off of the examples provided which all seemed to use `Tag<M,T>`. It was my assumption that this was how the library was meant to be used. I never used or looked at the base C library, so I didn't know what it offered.

Since I'm not familiar with libplctag itself, I guess what I'm not sure about is how I would do things without `Tag<M,T>`. Would we use just the Tag class and then just use its `Set[Type]` and `Get[Type]` methods like in [this example](https://github.com/libplctag/libplctag.NET/blob/%23406/examples/CSharp%20DotNetCore/ExampleSimple.cs)?

It seems I'd have to replace this code with something like a switch statement that just uses the base `Tag.Set[Type]` method based on the type I'm trying to read?

We're also doing a lot with arrays of tags and UDTs. That support is provided by the PlcMapper system, right? Otherwise we'd be looping through and reading the tag with these base Set and Get methods? I think the PlcMapper system is indeed preventing us from writing some of that code ourselves.

We're also using the code from the [`NotifyChanged`](https://github.com/libplctag/libplctag.NET/blob/%23406/examples/CSharp%20DotNetCore/PlcMapper/ExampleNotifyChanged.cs) example to keep things current. Is that something specific to `Tag<M,T>`?

I was somewhat watching the recent problems someone was having with Omron, is that part of what's prompting this? The differences between how the different PLC manufacturers handle the specifics and the mappers not handling things properly between this differences?

We do intend to support Omron as some of our customers do require their PLCs, though we haven't made any attempt to implement this so far. I was planning on attempting to use OPC UA (and that might be our path forward for all PLCs given the cybersecurity pushes for secure protocols that we're getting from some customers). It seems Rockwell is slow to adopt OPC UA and sounds like they want to charge extra for it, unlike Siemens who just include it in their CPUs.

timyhac, 2024-07-19:
Thanks @trinnan42 - this is great info!

> When I began using this a couple years ago. I based what I'm doing off of the examples provided which all seemed to use Tag<M,T>. It was my assumption that this was how the library was meant to be used.

100% correct.
Although [`Tag`](https://github.com/libplctag/libplctag.NET/blob/master/docs/libplctag.md#tag) has always been a type available to you, [`Tag<M,T>`](https://github.com/libplctag/libplctag.NET/blob/master/docs/libplctag.md#tagmt-and-mappers) was the one we used in all of the examples and I promoted.

> We're also doing a lot with arrays of tags and UDTs.

100% correct and this is where the Mapper system shines. Yes you would need to find some other way to do this.

> the PlcMapper system is indeed preventing us from writing some of that code ourselves.

100% correct.
There are some major limitations with the Mapper system:
* It does not allow you configure string attributes. Makes working with STRING types that don't have correct defaults impossible. You must use `Tag` instead. Many issues have been raised against this.
* It does not allow you to set up raw tags due to the need to Initialize the tag, then write to the buffer.
* Things like NotifyChanged are not supported directly (and can't be, due to the need to choose an appropriate hashing algorithm which the library can't do).
* `AutoSyncWriteInterval` [doesn't work](https://github.com/libplctag/libplctag.NET/issues/279).

Solving most of these issues would have been simpler if we instead used [inheritance](https://github.com/libplctag/libplctag.NET/blob/%23406/examples/CSharp%20DotNetCore/Inheritance/BaseTag.cs). In retrospect this would have been a better choice.
Having said that, it is not obvious what methods/properties the base class should expose, and it is trivial to write your own (particularly if there is an example to copy).

> I was somewhat watching the recent problems someone was having with Omron, is that part of what's prompting this?

It certainly is another example of the issue that library consumers face but we have had these problems for years.
For example, #212 has been open for almost 3 years.

> I'm concerned with needing more dependencies if the latter were the case. I would prefer that it remains part of the main library to needing another library to maintain.

This might not be the point you're making but potentially there is a case for remaining; in that you could argue being "official" it is more likely to get exposure, trust and momentum around it, and [therefore would be maintained at a higher level](https://en.wikipedia.org/wiki/Linus%27s_law).

> I create a dynamic tag variable by creating an instance of Tag<,> with the types from these dictionaries based on the Data Type Enum's value.

You can absolutely sidestep the Mapper system (even as libplctag.NET currently stands today) and do something like this:

```csharp
enum MyTagType { SINT, INT, DINT /* etc */ }

class MyTag
{
    private readonly Tag _tag;
    private Func<dynamic> decoder;
    private Action<dynamic> encoder;

    public MyTag(MyTagType tagType)
    {
        _tag = new Tag()
        {
            PlcType = PlcType.ControlLogix,
            Protocol = Protocol.ab_eip,
        };

        switch (tagType)
        {
            case MyTagType.SINT:
                _tag.ElementSize = 1;
                decoder = () => _tag.GetInt8(0);
                encoder = (value) => _tag.SetInt8(0, value);
                break;

            case MyTagType.INT:
                _tag.ElementSize = 2;
                decoder = () => _tag.GetInt16(0);
                encoder = (value) => _tag.SetInt16(0, value);
                break;

            case MyTagType.DINT:
                _tag.ElementSize = 4;
                decoder = () => _tag.GetInt32(0);
                encoder = (value) => _tag.SetInt32(0, value);
                break;
        }
    }

    public Task InitializeAsync() => _tag.InitializeAsync();
    public Task WriteAsync() => _tag.WriteAsync();
    public Task ReadAsync() => _tag.ReadAsync();

    public object Value { get => decoder(); set => encoder(value); }

    public string Gateway { get => _tag.Gateway; set => _tag.Gateway = value; }
    public string Path { get => _tag.Path; set => _tag.Path = value; }
    public string Name { get => _tag.Name; set => _tag.Name = value; }
}
```

Which would be used like this:

```csharp
var myTag = new MyTag(MyTagType.DINT)
{
    Name = "PROGRAM:SomeProgram.SomeDINT",
    Gateway = "10.10.10.10",
    Path = "1,0",
};

await myTag.ReadAsync();
var originalValue = myTag.Value;
Console.WriteLine($"Original value: {originalValue}");

int newValue = 3737;
myTag.Value = newValue;
await myTag.WriteAsync();
Console.WriteLine($"New value: {newValue}");
```

trinnan42, 2024-07-19:
@timyhac Your explanations are much appreciated! I've got a better understanding now. This certainly alleviates my concerns.

MountainKing91, 2024-07-20:
As a recent user of libplctag.NET, I too have to quote this:

> When I began using this a couple years ago. I based what I'm doing off of the examples provided which all seemed to use Tag<M,T>. It was my assumption that this was how the library was meant to be used.

Accordingly, I developed many custom mappers in order to work with all my custom datatypes and even for Omron system structure (_sAXIS_REF, etc.), Omron being the brand I use.

How I "share" tag values in my application is actually a mess, basically I register them in a list and access their values with a LINQ method finding them by name. I couldn't come up with a better idea at the time, and the I stayed with that.

Following this with interest.

timyhac, 2024-07-21:
Thanks for your input @MountainKing91!

MountainKing91, 2024-07-23:
> A study of applications or libraries that are consuming libplctag.NET shows that most will create yet another wrapper around Tag<M,T>.

@timyhac do you do this only by github search or is there anything else? I.e. user sharing their application or something?

timyhac, 2024-07-26:
@trinnan42, @MountainKing91 and @mitchrazga thankyou so much for your engagement and input!

I believe that removing the Mapper system and associated classes is still the right call to benefit the libplctag community long term, but have decided to phase out the mapper system in two stages to ease the transition:

1. Deprecate the mapper system, and rewrite examples and docs to use `Tag` instead of `Tag<M,T>`.
2. Remove the mapper system from the nuget library, only keep it as an example.

As you suggested, much of the issue is that the examples and documentation promote the use of the mapper system - so stage 1 will mitigate that part of the problem - while allowing current users to gradually transition.

For stage 2, I have not yet put in place an alternative for current users to migrate to, but I am thinking it will be a separate package but a drop-in replacement - it would be a separate package, and would not be the "official" .NET wrapper.

kyle-github, 2024-07-26:
Let me know your thoughts on how to set up a separate "extras" package.   I am really interested in doing that for the core DLL.  I want to pull out the examples, tools and tests (which are horribly mixed up now) from the main library.

timyhac, 2024-07-27:
Heya @kyle-github

The examples and tests would stay in the libplctag.NET github repository, but I was thinking that the classes shipped with the nuget package that I just [deprecated](https://github.com/libplctag/libplctag.NET/releases/tag/libplctag-v1.6.0-alpha.0) would be moved to a separate repository and released as a separate nuget package from there. The purpose being to make it clear that it is not a "libplctag" initiative.

A separate but related thought was that there could be a repository which _does_ attempt to document the behaviour of the various devices the libplctag community uses.
Its purpose would be to be a store of knowledge for the Modbus and Ethernet/IP services that those devices offer.
A non-goal would be to be comprehensive and to work out-of-the-box, but instead to document what the community has learnt so far, and to explicitly document where the knowledge gaps are.
This knowledge could be documented as executable programs - so that others could easily run the tests and confirm whether it is true on their device, or whether there is a gap in this knowledge that needs to be clarified.

MountainKing91, 2024-11-18:
Hi guys, I'm looking at the examples, trying to move past using the mappers.

I see in the examples folder that there are now two classes [BaseTag.cs](https://github.com/libplctag/libplctag.NET/blob/main/examples/CSharp%20DotNetCore/Inheritance/BaseTag.cs) and [Definitions.cs](https://github.com/libplctag/libplctag.NET/blob/main/examples/CSharp%20DotNetCore/Inheritance/Definitions.cs). Is this intended to be an alternative to mappers or am I missing something?
I will probably have the chance to start a new project with Omron + Libplctag, and I would like to start "fresh" and not use mappers if possible, even if until now I heavily relied on them.

Ty!

timyhac, 2024-11-18:
Yes exactly - in the top post I said a little bit about it under the "What's the alternative?" heading in the top post on this issue.

iancolledge, 2025-05-06:
For anyone googling moving away from mappers... ExampleListPlc.cs in the .NET Core examples folder is what you need.
We have about 30 or so mappers, they worked really well, but I can understand why they were deprecated. We'll re-write them when we get the chance. We did do a proof of concept a while back using our own low level routines in C# operating on the raw buffer avoiding the call to the C library and it was twice as fast basically, so we'll probably resurrect that code as part of this.

zmilliron, 2025-05-06:
Hi, new to this library and have been experimenting with it and saw that the intent was to deprecate the mapper system and that alternatives to it were being investigated it.  I'm currently working with ControlLogix and probably the easiest alternative is to just use marshalling.  If other platforms use fixed-size data types, then it's probably easily applicable to those as well.

I made some extension methods for the Tag class for illustration purposes that I'll share:

```csharp
public static T GetValue<T>(this Tag tag) where T : struct
{
    byte[] buffer = tag.GetBuffer();
    GCHandle handle = GCHandle.Alloc(buffer, GCHandleType.Pinned);

    try
    {
        //ToInt64 is used because it's assumed the code will compile and run on 64-bit platform.
        //If it's going to run on a 32-platform, ToInt32 should be used.
        T retVal = Marshal.PtrToStructure<T>(new IntPtr(handle.AddrOfPinnedObject().ToInt64()))!;

        return (retVal);
    }
    finally
    {
        handle.Free();
    }
}
```

The key to all of this is the `Marshal.PtrToStructure` method.  This will work with all atomic tags and convert them to .NET primitive types.  Also note that the method declaration limits T to structs so only .NET primitives and structs can be used.  Reference types (ie. classes) will not work.  Removing that restriction will throw an exception because Marshal.PtrToStructure does not accept reference types. 
 Sample usage:

```csharp
Tag t = new Tag()
{
    Name = "SomeDintTag"
    //additional properties
};

t.Read();
int i = t.GetValue<int>();
```

Even though strings are technically primitives in .NET, PtrToStructure will fail because strings aren't fixed length so it doesn't know what to do with them.  Additionally, strings in Logix are UDTs so they have to be handled specially anyways.  Which brings us to...

### Marshalling UDTs

You can marshal UDTs by declaring structures that implement the following rules:

* Structs must be decorated with a `StructLayoutAttrbute`
* LayoutKind must be set to **Sequential**
* The `Pack` property must be set to 4 because Logix sends UDTs in blocks of 4 bytes and pads out any blocks that are short.  Using Pack = 4 will make sure that padding is accounted for and each field is marshalled correctly.
* Fields in the struct must appear in the same order as fields in the UDTs and must have same size data types.

Take for example the following UDT

```
SomeUDT
    Field1 INT
    Field2 DINT
```
There's 6 total bytes, but Logix will send 8.  The first block will contain Field 1 in the first two bytes + two bytes of padding, and then Field 2 will be sent in the second block.  The corresponding struct is:

```csharp
[StructLayout(LayoutKind.Sequential, Pack = 4)]
public struct SomeUDT
{
    public short Field1;
    public int Field2;
}
```
No special accommodation needs to be made for padding, it's handled automatically. 

Logix strings can be converted with the following struct:
[... 298 lines cut ...]
public static void SetValue<T>(this Tag tag, T value) where T : struct
{
    byte[] buffer = new byte[Marshal.SizeOf<T>()];
    GCHandle handle = GCHandle.Alloc(buffer, GCHandleType.Pinned);
    try
    {
        //ToInt64 assuming 64-bit platform
        Marshal.StructureToPtr(value!, new IntPtr(handle.AddrOfPinnedObject().ToInt64()), false);
        tag.SetBuffer(buffer);
    }
    finally
    {
        handle.Free();
    }
}
```
I'll admit that I haven't been able to extensively test writing.  I've only been able to access PLCs that are "live," so I don't have one to play with where I can write arbitrary values and see what happens.  However, I've been able to test and see that the byte buffer array appears to be set correctly.  Usage is to just call the Set* method with your data type and then tag.Write();

Existing code in the libplctag library allows reading and writing to offsets, and that should be something easily integrated into these extension methods.  They ultimately map to addresses, so an offset parameter could be added to the address before marshalling.

Finally, I'll also point out I've done no performance benchmarking.  I don't know how well this works with large volumes of data or a large number of read/writes.  For my purposes it's performed as well as it's needed to.

### Misc. Notes
* This will NOT work with @tags since tag names are returned as variable-length strings instead of UDT-style fixed-length strings.
* You can declare structs with 'LayoutKind.Explicit', but you then have to decorate every field with a `FieldOffsetAttribute` and you can't use the `Pack` property and instead have to meticulously calculate every field offset in the buffer.  The tradeoff is more work for the flexibility to order the fields any way you want in your struct definition.

timyhac, 2025-05-08:
Hi @zmilliron - this is a fantastic contribution, thanks so much for sharing!

If you're up for it, it would be great to add your marshalling concept as another alternative in the [Examples](https://github.com/libplctag/libplctag.NET/tree/main/examples/CSharp%20DotNetCore).

sebastian-zitzelsberger, 2025-09-17:
@timyhac  Is there a reason why [this constructor of the Tag class](https://github.com/libplctag/libplctag.NET/blob/20480ebec6cffd2b45adccd1778f8260da642476/src/libplctag/Tag.cs#L71)  was made internal? I would like to write test for custom Tag-classs that use the real Tag class.
And then pass a mock for the INative for local testing without a device for both the tag +  any encoding or decoding.
But as the constructor is internal and the class no longer has an interface. This is now unnecessary complicated because the design ingores the open-closed principal.

alessandrossi97, 2026-08-18:
Late to this thread, and apologies if it's out of scope given how long it's been quiet. I'm only offering this in case it's of any help in improving the libraries — feel free to ignore it if it isn't.

A data point in favour of removing the Mapper system, from the perspective of someone who wraps `libplctag.NativeImport` directly rather than consuming the `libplctag` .NET package and `Tag<M,T>`.

**Context.** I maintain a small family of PLC libraries: one per protocol family (Rockwell over `NativeImport`, plus others for different vendors), all sharing a common base library that handles the parts that aren't protocol-specific. The Rockwell wrapper polls cyclically — atomic tags are read as atomic tags, UDT instances are read as raw bytes — and then distributes the decoded values onto the individual atomic tags it exposes to the application.

**First iteration: code generation (mapper-shaped).** I wrote a generator that, from a tag listing, produced the UDT classes, a mapper per UDT type, and a static class holding the UDT type ids as constants. UDTs were then read through `Decode`. It worked, but it had two costs I hadn't anticipated:

- every project had to derive from my wrapper class just to make it aware of its own generated types;
- any change on the PLC side meant regenerating the UDT classes, mappers and id constants, then redeploying.

I also never used `Encode`. Assembling a whole struct in memory in order to write a single member felt like a good way to introduce silent mapping errors, so writes are done by creating one atomic tag, writing it, and disposing it.

**Second iteration: runtime metadata.** Now the wrapper browses the controller on connect — `@tags`, then `@udt/{id}` for each distinct UDT id — and caches a dictionary of UDT id to UDT definition in memory. On every poll cycle it already knows the layout of every structure, so it can slice the raw buffer and assign each member to the correct atomic tag with no generated code at all. If the PLC program changes, I restart the service and it picks up the new layout. The same library now runs unmodified across every project in my environment.

<details>
<summary>Sketch of the decode path (simplified from my wrapper)</summary>

This is trimmed down for readability — diagnostics and a few library-internal helpers are elided, and `LookupTag` stands in for however you resolve an address to an application-level tag object.

```csharp
public void ReadUdt(ITag tag, byte[] buffer, int offset, UdtInfo info, string newAddress = "")
{
    // Top-level entry point: no buffer yet, so read the whole UDT instance as raw bytes.
    if (buffer == null)
    {
        tag.Read();
        buffer = new byte[info.Size];
        plc_tag_get_raw_bytes(tag.Handle, 0, buffer, buffer.Length);
    }

    var baseAddress = string.IsNullOrEmpty(newAddress) ? tag.Address : newAddress;

    foreach (var field in info.Fields)
    {
        // Padding and host-generated members are not exposed to the application.
        if (IsExcludedField(field.Name))
            continue;

        var address = $"{baseAddress}.{field.Name}";

        if (IsAtomicType(field.Type))
        {
            if (field.IsArray && field.ArrayLength > 0)
            {
                if (IsBoolArray(field.Type))
                {
                    // Handles booleans.
                    var elementIndex = 0;
                    for (int byteIndex = 0; byteIndex < field.ArrayLength / 8; byteIndex++)
                    {
                        for (int bitIndex = 0; bitIndex < 8; bitIndex++)
                        {
                            var target = LookupTag($"{address}[{elementIndex}]");
                            if (target != null)
                            {
                                target.SetValue(
                                    buffer.GetObjectValue(this, tag.Handle,
                                        offset + (int)field.Offset + byteIndex,
                                        bitIndex,
                                        field.Type));
                            }
                            elementIndex++;
                        }
                    }
                }
                else
                {
                    var elementSize = GetObjectSize(field.Type);
                    for (int i = 0; i < field.ArrayLength; i++)
[... 57 lines cut ...]
    }
}

public UdtInfo GetUdt(int typeId)
{
    // UDT definitions are immutable for the lifetime of the connection,
    // so they are fetched once and cached.
    var id = (ushort)typeId;

    if (_udtCache.TryGetValue(id, out var cached))
        return cached;

    var definition = ReadUdtDefinition($"@udt/{id}");
    _udtCache.Add(id, definition);
    return definition;
}
```

</details>

Two caveats worth stating plainly. The browse-on-connect step needs to be evaluated per application: in my case the startup cost is acceptable, but on a large PLC project it could introduce a noticeable delay before the first poll cycle, and anyone adopting this pattern should size that up front. And I have only ever tested this against ControlLogix — I have no data on how it behaves on Omron or the other supported families, which is precisely the vendor-differences problem you raise above.

The point being: what made this possible was direct access to the tag buffer plus the controller's own metadata — not a mapping layer. The decoding logic in `examples/CSharp DotNetCore/ExampleListPlc.cs` was ef [... cut ...] ve accessors. It also seems worth noting for one of the bullets in your summary: for Logix UDTs at least, the controller can describe its own tag buffer encoding at runtime, which reduces (though doesn't remove) the need for a static registry of known encodings.

So: +1 to removing the Mappers from the official package. If a worked example along these lines — browse on connect, cache the UDT definitions, decode generically — would be useful in the Examples folder, I'd be happy to contribute one.

timyhac, 2026-11-15:
This change has been made in https://github.com/libplctag/libplctag.NET/releases#release-libplctag-v2.0.0-alpha.2, the classes in question were marked obsolete in https://github.com/libplctag/libplctag.NET/releases#release-libplctag-v1.6.0-rc.1

closed 2026-11-15 by timyhac