libplctag/libplctag.NET #409: Net Framework 4.5

t1a2l, 2024-07-30:
it seems it is required to have 3.1, i have a working project with 4.5 and changing it back down is not an option, any solution, are you going to support 4.5?

timyhac, 2024-07-30:
Hi there - I think what you are saying is that you

* have a project which targets .NET Framework 4.5 and uses the [libplctag Nuget package.](https://www.nuget.org/packages/libplctag).
* have recently updated from libplctag-v1.5.2 to libplctag-v1.6.0-alpha.0
* have found that the libplctag-v1.6.0-alpha.0 does not target NET Framework 4.5

---

> Changing it down is not an option

.NETCore 3.1 is more recent than .NET Framework 4.5 - both of these have reached End of Life by Microsoft.

There are multiple versions of ".NET" - the two that are usually discussed are
* .NET Framework
* .NET Core (and .NET)

.NET Framework 4.5 has not been supported by Microsoft [since 2016](https://learn.microsoft.com/en-us/lifecycle/products/microsoft-net-framework), whereas .NETCore 3.1 was only released in December 2019 and reached end of life Dec 2022.
.NETCore was rebranded to .NET when moving from 3.1 to 5 - so you will see it as NETCore 3.1, and NET5.0

---

> are you going to support 4.5?

The project has always (up until the recent libplctag-v1.6.0-alpha.0) targeted NETStandard2.0, which is able to be used by [certain versions](https://learn.microsoft.com/en-us/dotnet/standard/net-standard?tabs=net-standard-2-0) of NETFramework in addition to NETCore and .NET and NETFramework 4.5 is not one of them.

So if my assumption is correct that you have been successfully using libplctag-v1.5.2 for a NETFramework4.5 project - I'm not sure how that has been working.

Having said that, if there is a good reason to add support for NetFramework4.5 then I am open to that but I believe there would be additional testing capabilities which I currently don't have easy access to.

MountainKing91, 2024-08-03:
In hope it helps somehow, all my recent tests and projects with libplctag are developed using .NET Framework 4.8, which is supported by .NET Standard 2.0 as @timyhac's link states and these days it is also already installed on Windows 10 by default - this comes handy.

@t1a2l  Is there any chance you can upgrade from 4.5 to 4.8?

timyhac, 2024-08-06:
@t1a2l - I trust that this has answered your question, and I agree with @MountainKing91 - upgrading to a supported version of .NET is preferable. If you do have a good reason to add supported for end-of-life versions of .NET please let us know.

closed 2024-08-06 by timyhac