libplctag/libplctag.NET #412: Support to MAUI

ElegantCatterpillar, 2024-08-24:
Im trying to used in MAUI and get this error:

System.TypeLoadException: Unknown platform
   at libplctag.NativeImport.LibraryExtractor.GetAppropriateLibraryInfo() in /_/src/libplctag.NativeImport/LibraryExtractor.cs:line 90
   at libplctag.NativeImport.LibraryExtractor.ExtractAppropriateLibraryToDirectory(String outputDirectory) in /_/src/libplctag.NativeImport/LibraryExtractor.cs:line 43
   at libplctag.NativeImport.LibraryExtractor.Init(Boolean forceExtract) in /_/src/libplctag.NativeImport/LibraryExtractor.cs:line 25
   at libplctag.NativeImport.plctag.ExtractLibraryIfRequired() in /_/src/libplctag.NativeImport/plctag.cs:line 59
   at libplctag.NativeImport.plctag.plc_tag_create_ex(String lpString, callback_func_ex func, IntPtr userdata, Int32 timeout) in /_/src/libplctag.NativeImport/plctag.cs:line 94
   at libplctag.NativeTag.plc_tag_create_ex(String lpString, callback_func_ex func, IntPtr userdata, Int32 timeout) in /_/src/libplctag/NativeTag.cs:line 20
   at libplctag.NativeTagWrapper.InitializeAsync(CancellationToken token) in /_/src/libplctag/NativeTagWrapper.cs:line 406
   at libplctag.Tag`2.<InitializeAsync>d__46[[libplctag.DataTypes.BoolPlcMapper, libplctag, Version=1.5.2.0, Culture=neutral, PublicKeyToken=null],[System.Boolean, System.Private.CoreLib, Version=8.0.0.0, Culture=neutral, PublicKeyToken=7cec85d7bea7798e]].MoveNext() in /_/src/libplctag/TagOfT.cs:line 169

timyhac, 2024-08-24:
It works on my machine - if I create a new MAUI app and just run it will all default settings, it works fine 😛

But in all seriousness there is a lot more detail that you could be adding.
MAUI is a cross-platform framework so I could take a guess that you are running this on something other than windows, macOS or linux which are the version of libplctag core that are bundled with [libplctag.NET](https://github.com/libplctag/libplctag.NET/tree/main/src/libplctag.NativeImport/runtimes).

Please see the [CONTRIBUTING guidelines](https://github.com/libplctag/libplctag.NET/blob/main/CONTRIBUTING.md) for additional detail to add.

ElegantCatterpillar, 2024-08-25:
I am trying to use it on Android, but I get an error from the library that it cannot Extract the library, specifically the error I get is the one at the top which mentions that it does not recognize the platform it is running on.