libplctag/libplctag.NET #413: Reading CPU state

johnvarney612, 2024-08-29:
Moving code using Net.Logix to use libPlcTag.NET instead.
Net.Logix has two special tags $CPU_STATE and $CPU_INFO that access system data usually accessed in the PLC program using GSV eg accessing the ControllerDevice attributes.
I have found info to use @raw to read the wall clock time but no examples to read other class/attribute combinations.
@tags does not list the system tag data.
The rawPayload for reading the wall clock time is not obvious. 
Any help would be useful.
Thanks

kyle-github, 2024-08-29:
Hmm, we should probably set up a wiki page with common class/instance/attribute data for raw tags to use to get/set common things like time and these CPU states.   

Any chance you can use Wireshark and capture what the Net.Logix library does when it gets those special tags?

johnvarney612, 2024-08-29:
I'll look into getting the data.
Thanks for the very fast response

timyhac, 2024-10-08:
Hi @johnvarney612 - have you had a chance to capture the Wireshark logs for this?

timyhac, 2025-01-26:
Hi @johnvarney612 - just a reminder, did you end up getting a chance to capture the wireshark logs for this?

timyhac, 2025-07-20:
Hi @johnvarney612 - if you'd still like to grab this it would be great to provide the above info. Closing this issue for now.