libplctag/libplctag.NET #416: MicroLogix 1400 : plc_tag_create return PLCTAG_ERR_BAD_PARAM

DamienPim, 2024-09-12:
Hello, I can't get libplctag to work with MicroLogix 1400.
I use libplctag 1.5.2 and libplctag.NativeImport in a VS2022 C# project.
I call plc_tag_create with following UniqueKey:
"protocol=ab_eip&gateway=192.168.0.98&cpu=MicroLogix&elem_size=1&elem_count=1&name=Bit_1"
but I get PLCTAG_ERR_BAD_PARAM error.
For cpu, I see that I can use MicroLogix because in get_plc_type, stricmp is used to compare cputypes...
Can anyone help me?
Has anyone tried libplctag with a MicroLogix 1400 ?
Thank you in advance.

timyhac, 2024-09-12:
Searching existing issues for "MicroLogix 1400" yields that someone has had some success already: https://github.com/search?q=repo%3Alibplctag%2Flibplctag.NET+micrologix+1400&type=issues

Can you please turn on [debugging to level 4](https://github.com/libplctag/libplctag/wiki/API#debugging), and provide other information such as [a code snippet of what you have tried, and let us know if you have been able to get _something_ working, even if it isn't the final result you're looking for.](https://github.com/libplctag/libplctag.NET/blob/main/CONTRIBUTING.md#guidance-when-creating-an-issue)

Unpacking what you are saying:

* Your application is a C# .NET Core application (which framework version?)
* Both the libplctag package (v1.5.2) and the libplctag.NativeImport package (only as a transitive reference? If not, which version?) have been added to the application.
* Within your code you have a call to [`plc_tag_create`](https://github.com/libplctag/libplctag/wiki/API#creating-a-tag-handle) and you pass in an attribute string of `"protocol=ab_eip&gateway=192.168.0.98&cpu=MicroLogix&elem_size=1&elem_count=1&name=Bit_1"`.
* You want to be able to write `cpu=MicroLogix` instead of [`cpu=micrologix`](https://github.com/libplctag/libplctag/wiki/Tag-String-Attributes#ab-specific-attributes).

DamienPim, 2024-09-12:
I don't specify path because I see that it is not used for MicroLogix PLC.
I write MicroLogix and not micrologix because I see a case insensitive comparison.
Both the libplctag package (v1.5.2) and the libplctag.NativeImport package (v1.0.41) have been added to the application.
C# .Net Framework 4.6.2. Tests on Windows 10.
The MicroLogix1400 on which I can make tests will be no longer available after this afternoon. I have to find another MicroLogix PLC to continue my tests...
Thank again for your help.

kyle-github, 2024-09-12:
`name=Bit_1` is not a valid DF1/PCCC data file reference.   `B42:0/1` is.   

That said, the debug file will help a lot.

timyhac, 2025-07-20:
G'day @DamienPim - were you able to get the debug logs in the end?

kyle-github, 2025-07-25:
For Micrologix you need to use the D1/PCCC data file names.   N7:0.  B3:2/1 etc.  I have a Micrologix 1100 with Ethernet and it works.  I know, I know, "works for me" is not that helpful :-/

timyhac, 2025-07-26:
@DamienPim - thanks for the response, if you do encounter further issues with libplctag or libplctag.NET in the future please don't hesitate to reach out to the community.

closed 2025-07-26 by timyhac