libplctag/libplctag.NET #418: Callback seems to remain on failed InitializeAsync

dnmalenke, 2024-10-02:
When initializing a tag that will fail with ErrorNotFound, it seems like native callbacks are registered, but not disposable. This leads to an application crash/hang on `LibPlcTag.Shutdown()`.
Log entry:
```
15:56:32 VRBProcess terminated. A callback was made on a garbage collected delegate of type 
 libplctag.NativeImport!libplctag.NativeImport.plctag+callback_func_ex::Invoke'.
--------------------------------
   at libplctag.NativeImport.NativeMethods.plc_tag_shutdown()
--------------------------------
   at libplctag.NativeImport.plctag.plc_tag_shutdown()
   at libplctag.NativeTag.plc_tag_shutdown()
   at **libplctag.LibPlcTag.Shutdown()**
```

Unfortunately, I was unable to replicate this on a smaller scale. 

I am able to workaround the issue by calling `.Initialize()` instead of `.InitializeAsync()` or forcing the `_isInitialized` field to true via messy reflection before disposing:
```
 try
 {
     await tag.InitializeAsync();
 }
 catch
 {
     var foundF = typeof(Tag).GetField("_tag", System.Reflection.BindingFlags.NonPublic | System.Reflection.BindingFlags.Instance);
     var pTag = foundF!.GetValue(tag);
     pTag.GetType().GetField("_isInitialized", System.Reflection.BindingFlags.NonPublic | System.Reflection.BindingFlags.Instance).SetValue(pTag, true);
     tag.Dispose();

     throw;
 }
 ```

This forces the Dispose call to unregister callbacks and destroy the handle.

Any ideas?
Thanks

timyhac, 2024-10-03:
I havent had a chance to sit down and review this but yes youre probably
onto something. Probably need to unregister the callbacks here:
https://github.com/libplctag/libplctag.NET/blob/ae44f94713b379591a1b20f274c8986f59ab63ae/src/libplctag/Tag.cs#L706

timyhac, 2024-10-08:
@dnmalenke - although it makes sense that the callbacks need to be removed, I was not able to trigger the issue with a unit test.

It also doesn't make sense that the issue _wouldn't_ occur when using `Initialize` - the same callbacks are added.

Having said all this I think it still is a problem so have put together PR #420 - is it possible you could test it out to see if it resolves the problem?

dnmalenke, 2024-10-08:
@timyhac Thank you, this seems to have resolved the issue. 

I do want to note, that with this update, the function `public void GetBuffer(byte[] buffer)` stopped working. I had to use `public byte[] GetBuffer()` within my decoding functions instead

timyhac, 2024-10-08:
Thanks for testing.

I'm not sure that the GetBuffer issue is related to this specific change -
could you raise a separate issue with the normal details like a repro,
debug logs, versions, etc

timyhac, 2024-12-01:
Resolved in libplctag.NativeImport-v2.0.0-alpha.5

closed 2024-12-01 by timyhac