libplctag/libplctag.NET #424: Trying to read compactlogix l30er getting a CIP Error code port not available

Buzznacker, 2024-10-21:
I am using libplctag 1.2.1

Trying to read a compactlogix l30er and I keep getting CIP Error code port not available.
I've tried using Telnet on the 44818 port on the address of the PLC and I am able to successfully connect so I know the PLC is accessible.

```
var tag = new TagBool
{
    Gateway = ip,
    Path = "1,0",
    Name = tagName,
    Protocol = Protocol.ab_eip,
    PlcType = PlcType.ControlLogix,
    DebugLevel = DebugLevel.Detail
};
```

> 2024-10-21 18:41:32.417 thread(3) tag(0) DETAIL session_handler:1129 Send Forward Open succeeded, going to SESSION_RECEIVE_FORWARD_OPEN state.
2024-10-21 18:41:32.417 thread(3) tag(0) DETAIL session_handler:1136 in SESSION_RECEIVE_FORWARD_OPEN state.
2024-10-21 18:41:32.417 thread(3) tag(0) INFO receive_forward_open_response:2323 Starting
2024-10-21 18:41:32.417 thread(3) tag(0) INFO recv_eip_response:1946 Starting.
2024-10-21 18:41:32.417 thread(3) tag(0) DETAIL socket_read:1754 Starting.
2024-10-21 18:41:32.417 thread(3) tag(0) DETAIL socket_read:1786 Immediate read attempt did not succeed, now wait for select().
2024-10-21 18:41:32.419 thread(3) tag(0) DETAIL socket_read:1814 Socket can read data.
2024-10-21 18:41:32.419 thread(3) tag(0) DETAIL socket_read:1866 Done: result 24.
2024-10-21 18:41:32.419 thread(3) tag(0) DETAIL socket_read:1754 Starting.
2024-10-21 18:41:32.419 thread(3) tag(0) DETAIL socket_read:1866 Done: result 38.
2024-10-21 18:41:32.419 thread(3) tag(0) INFO recv_eip_response:2017 request received all needed data (62 bytes of 62).
2024-10-21 18:41:32.419 thread(3) tag(0) INFO recv_eip_response:2019 00000 6f 00 26 00 08 00 ee ee 00 00 00 00 8c 93 6c 50
2024-10-21 18:41:32.419 thread(3) tag(0) INFO recv_eip_response:2019 00016 00 00 00 00 00 00 00 00 00 00 00 00 00 00 02 00
2024-10-21 18:41:32.419 thread(3) tag(0) INFO recv_eip_response:2019 00032 00 00 00 00 b2 00 16 00 d4 00 01 01 11 03 c8 ba
2024-10-21 18:41:32.419 thread(3) tag(0) INFO recv_eip_response:2019 00048 3d f3 45 43 50 21 03 00 01 00 20 02 24 01
2024-10-21 18:41:32.419 thread(3) tag(0) INFO recv_eip_response:2026 Done.
2024-10-21 18:41:32.419 thread(3) tag(0) WARN receive_forward_open_response:2347 Forward Open command failed, response code: Port Not Available (1)
2024-10-21 18:41:32.419 thread(3) tag(0) WARN receive_forward_open_response:2372 CIP error code Port Not Available (This error occurs when a port number specified in a port segment is not available. Consult the documentation of the device to verify the correct port number.)!
2024-10-21 18:41:32.419 thread(3) tag(0) INFO receive_forward_open_response:2390 Done.
2024-10-21 18:41:32.419 thread(3) tag(0) WARN session_handler:1151 Receive Forward Open failed PLCTAG_ERR_REMOTE_ERR!

kyle-github, 2024-10-21:
Try a path of "0,0".   IIRC, that is correct for those PLCs.  I do not have one to test on so I am not sure.

Buzznacker, 2024-10-21:
@kyle-github 
> Try a path of "0,0". IIRC, that is correct for those PLCs. I do not have one to test on so I am not sure.

Thanks for your reply, I've tried this and still no changes still getting the CIP Error

timyhac, 2025-01-25:
I used to use a similar model at a previous workplace and "1,0" was what I used.

Were you able to resolve this in the end?

timyhac, 2025-07-20:
Hi @Buzznacker - if this is still a problem, please let us know. Probably the main repo is the right place as this isn't a wrapper-specific problem.

closed 2025-07-20 by timyhac