libplctag/libplctag.NET #427: Read LINT as BOOL array

alessandrossi97, 2025-01-08:
Hi everyone,

Is there a way to read integer types (INT, DINT, LINT) as boolean arrays?
I am using the following code:

```
var bit = new Tag<BoolPlcMapper, bool>()
{
    Name = "ALARMS_GENERAL.SeriuosAlarms.0",
    Gateway = "172.27.7.205",
    Path = "1,0",
    PlcType = PlcType.ControlLogix,
    Protocol = Protocol.ab_eip,
    Timeout = TimeSpan.FromSeconds(5)
};
Console.WriteLine($"{bit.Name} value: {bit.Read()}.");

var array = new Tag<BoolPlcMapper, bool[]>()
{
    Name = $"ALARMS_GENERAL.SeriuosAlarms",
    Gateway = "172.27.7.205",
    Path = "1,0",
    ArrayDimensions = new int[] { 64 },
    PlcType = PlcType.ControlLogix,
    Protocol = Protocol.ab_eip,
    Timeout = TimeSpan.FromSeconds(5)
};
Console.WriteLine($"{array.Name} value: {array.Read()}.");
```

ALARMS_GENERAL.SeriuosAlarms is a LINT variable, I can correctly read single bits, but i cannot read the LINT as bool array. The array initializer method raises an exception and returns _ErrorOutOfBounds_.

Any help would be greatly appreciated,
Thanks in advance

kyle-github, 2025-01-08:
Within the PLC and within the CIP protocol, data access to tags is typed and the type of the access must match the type of the tag.   That said, you can use the [bit accessor](https://github.com/libplctag/libplctag/wiki/API#bit-accessors) functions (change to camel-case for C#/.NET I think) to get individual bits.   You could write your own mapper with that.

timyhac, 2025-01-25:
I trust this was the information you needed.

If you need some inspiration on Mappers, there are some examples to study [here](https://github.com/libplctag/libplctag.NET/tree/main/examples/CSharp%20DotNetCore/PlcMapper).

closed 2025-01-25 by timyhac