libplctag/libplctag.NET #428: Error writing to string in Omron PLC

VitorKawao, 2025-01-10:
Hello,

I am having some issues trying to write a string to an Omron (Nx1p2) PLC. I can read correctly, and I can write as long as I do not change the size of the string. If my tag has 3 characters, I can only write another word with 3 characters; it cannot be 2 or 4 characters.

My code
```
Tag tag = new Tag()
{
    Gateway = ip,
    Path = "18," + ip,
    PlcType = PlcType.Omron,
    Name = tagName,
    ElementSize = 256,
    Protocol = Protocol.ab_eip,
    Timeout = TimeSpan.FromMilliseconds(1000),

    StringIsCounted = true,
    StringIsFixedLength = false,
    StringIsZeroTerminated = false,
    StringCountWordBytes = 2,
    StringPadBytes = 0,        
};

tag.Read();
string sValue = tag.GetString(0);
Console.WriteLine("Value: " + sValue);

sValue += "a";    
try
{
    //tag.SetSize(sValue.Length);
    tag.SetString(0, sValue);
    tag.Write();
}
catch (Exception e)
{
    Console.WriteLine(e.Message);
    Console.WriteLine(e.StackTrace);
}
```

The exception  is ErrorNotAllowed
[image]

I would like to know if I am doing something wrong.

Thank you,
Vítor Guedes

timyhac, 2025-01-10:
Hi - thanks for the information. It would be good if you could also provide [debug logs and other helpful information](https://github.com/libplctag/libplctag.NET/blob/main/CONTRIBUTING.md#guidance-when-creating-an-issue).

You have commented out the resizing of the buffer which you do need to do, though the value you need to [provide](https://github.com/libplctag/libplctag/wiki/API#retrievingsetting-tag-size) is for the entire buffer size, not just the character data.

VitorKawao, 2025-01-13:
Hello,

Regarding the setting of tag size, I tried using it as well (tag.SetSize(sValue.Length), but the error is the same.

I am attaching my  debug logs
[log.txt](https://github.com/user-attachments/files/18397672/log.txt)

Thank you,
Vítor Guedes

kyle-github, 2025-01-13:
You can try to use the [allow_field_resize](https://github.com/libplctag/libplctag/wiki/All-Attributes#allow_field_resize) attribute to see if that works.  However, you must have all the other string definition attributes set as well.

kyle-github, 2025-01-13:
If you set that attribute, then just using [plc_tag_set_string()](https://github.com/libplctag/libplctag/wiki/API#writing-a-string) might work.

VitorKawao, 2025-01-14:
Hello,

How do I change this attributte at c#?
I cannot see a AllowFieldResize property at Tag.cs

timyhac, 2025-01-14:
> I tried using it as well (tag.SetSize(sValue.Length), but the error is the same.

The size of the buffer is the length of the string plus the word count, in your case you set this to 2. You can use `GetBuffer()` after you initialize or read to inspect the buffer content.

> How do I change this (AllowFieldResize) attributte at c#?

I've added this in [libplctag-v1.6.0-alpha.2](https://github.com/libplctag/libplctag.NET/releases/tag/libplctag-v1.6.0-alpha.2)

VitorKawao, 2025-01-15:
After migrating to libplctag-v1.6.0-alpha.2 and setting AllowFieldResize to true, it works correctly

Thank you,
Vítor Guedes

closed 2025-01-15 by timyhac