libplctag/libplctag.NET #429: Issue while writing custom string tag on to plc [error bad config].

ArunPusalapati, 2025-01-10:
When I try to run the code I am getting exception as Error bad config. I have a string restriction as 32 characters.

var stringTag1 = new Tag()
{

    Name = "TestString32",
    Gateway = gateway,
    Path = path,
    PlcType = PlcType.ControlLogix,
    Protocol = Protocol.ab_eip,
    ElementSize=32,
    ElementCount = 1,
    Timeout = TimeSpan.FromSeconds(5)

};
stringTag1.Initialize();
stringTag1.setString(0,"test")
stringTag1.write();
stringTag1.Read();

timyhac, 2025-01-10:
Reading the [manual on Strings](https://github.com/libplctag/libplctag/wiki/Strings-and-String-Handling) you'll see that additional configuration for the Tag is required.

If you need additional help could you please check first through [these guidelines](https://github.com/libplctag/libplctag.NET/blob/main/CONTRIBUTING.md#guidance-when-creating-an-issue).

timyhac, 2025-01-25:
I trust that you were able to get the information you needed from the link.

closed 2025-01-25 by timyhac