libplctag/libplctag.NET #437: Cannot Declare Array[0..15] Bool

zN3utr4l, 2025-01-27:
```C#
tag = new Tag<M, T>
{
    Name = itemModel.ItemID,

    Gateway = configuration.ServerIp,

    Path = configuration.ServerPath,

    PlcType = configuration.PlcType == Enums.EEIPPlcType.CompactLogix ? PlcType.ControlLogix : (PlcType)configuration.PlcType,

    Protocol = configuration.Protocol,

    Timeout = TimeSpan.FromMilliseconds(configuration.ItemTimeOut),
};

if (itemModel.IsArray)
{
    tag.ArrayDimensions = [(int)itemModel.Dimension];
}

tag.Initialize();
```

`<VariableArray itemId="Allarms" dataType="BOOL" dimension="16" />`

I have this tag, a bool array of 16 positions, which when I try to monitor gives me the error: OutOfBound, but if I declare the array of size 32 at plc level and go to monitor the 16, as before, it works.

kyle-github, 2025-01-27:
I forgot to mention that you need to set the number of elements in the tag (in the .NET code creating the tag) to the number of UDINT elements, NOT the number of BOOL elements.

timyhac, 2025-01-27:
We attempted to encode this logic into a [BoolMapper](https://github.com/libplctag/libplctag.NET/blob/main/src/libplctag/DataTypes/BoolPlcMapper.cs#L31), but have found it only works in some very specific scenarios. Bool arrays were one of the reasons I decided to [deprecate (and eventually remove) the Mapper system entirely](https://github.com/libplctag/libplctag.NET/issues/406).

In terms of debugging this following Kyle's advice you could create a `Tag` as follows:

```csharp
Tag tag = new()
{
    Name = "Allarms",
    Gateway = ...,
    Path = ...,
    PlcType = PlcType.ControlLogix,
    Protocol = Protocol.ab_eip,
    ElementSize = 4,
    ElementCount = 1,
};

tag.Initialize();
tag.SetBit(3, true);       // Sets "Allarms[3]" to true
tag.Write();
```

If you want to continue to use the Mapper system, you will need to code test this for all scenarios you expect to support - I see that you're allowing for multiple Protocols and PlcTypes in your code, so you'll need to determine how each combination encodes BOOL arrays and code for that.

zN3utr4l, 2025-01-28:
Exactly, I have some generic functions that initialize the Tag:

```C#
protected internal IItem? GetArrayItem(EEIPItemModel itemModel)
{
    return itemModel.DataType switch
    {
        ItemDataType.SINT => InitializeTag<SintPlcMapper, sbyte[]>(itemModel),

        ItemDataType.USINT => InitializeTag<BytePlcMapper, byte[]>(itemModel),

        ItemDataType.INT16 => InitializeTag<IntPlcMapper, short[]>(itemModel),

        ItemDataType.DINT => InitializeTag<DintPlcMapper, int[]>(itemModel),

        ItemDataType.LINT => InitializeTag<LintPlcMapper, long[]>(itemModel),

        ItemDataType.WORD => InitializeTag<UShortPlcMapper, ushort[]>(itemModel),

        ItemDataType.DWORD => InitializeTag<UIntPlcMapper, uint[]>(itemModel),

        ItemDataType.ULINT => InitializeTag<ULongPlcMapper, ulong[]>(itemModel),

        ItemDataType.REAL => InitializeTag<RealPlcMapper, float[]>(itemModel),

        ItemDataType.LREAL => InitializeTag<LrealPlcMapper, double[]>(itemModel),

        ItemDataType.BOOL => configuration.PlcType == EEIPPlcType.CompactLogix
                                                            ? InitializeTag<ExtesionClass.BoolPlcMapper, bool[]>(itemModel)
                                                            : InitializeTag<libplctag.DataTypes.BoolPlcMapper, bool[]>(itemModel),

        ItemDataType.STRING => InitializeTag<ExtesionClass.StringPlcMapper, string[]>(itemModel),

        _ => null,
    };
}

protected internal IItem? GetItem(EEIPItemModel itemModel)
{
    return itemModel.DataType switch
    {
        ItemDataType.SINT => InitializeTag<SintPlcMapper, sbyte>(itemModel),

        ItemDataType.USINT => InitializeTag<BytePlcMapper, byte>(itemModel),

        ItemDataType.INT16 => InitializeTag<IntPlcMapper, short>(itemModel),

        ItemDataType.DINT => InitializeTag<DintPlcMapper, int>(itemModel),

        ItemDataType.LINT => InitializeTag<LintPlcMapper, long>(itemModel),

        ItemDataType.WORD => InitializeTag<UShortPlcMapper, ushort>(itemModel),

        ItemDataType.DWORD => InitializeTag<UIntPlcMapper, uint>(itemModel),

        ItemDataType.ULINT => InitializeTag<ULongPlcMapper, ulong>(itemModel),

        ItemDataType.REAL => InitializeTag<RealPlcMapper, float>(itemModel),

        ItemDataType.LREAL => InitializeTag<LrealPlcMapper, double>(itemModel),

        ItemDataType.BOOL => configuration.PlcType == EEIPPlcType.CompactLogix
                                                            ? InitializeTag<ExtesionClass.BoolPlcMapper, bool>(itemModel)
                                                            : InitializeTag<libplctag.DataTypes.BoolPlcMapper, bool>(itemModel),

        ItemDataType.STRING => InitializeTag<ExtesionClass.StringPlcMapper, string>(itemModel),

        _ => null,
    };
}

protected internal IItem? InitializeTag<M, T>(EEIPItemModel itemModel) where M : IPlcMapper<T>, new()
{
    try
    {
        return new EEIPItem<M, T>(itemModel, configuration);
    }
    catch (LibPlcTagException ex)
    {
        throw new ItemException(MessagesError.EEIPItemCreateException, ex);
    }
}
```

new EEIPItem() performs the tag initialization action I mentioned in the first comment.
So in practice this generic logic is not good ? because ElementSize of each mapper should change for each plc?

Some of the mappers I had to create myself (Bool, Byte, String, UInt, ULong, UShort)

kyle-github, 2025-01-28:
Be careful.  A single BOOL is returned as a byte value (IIRC).   But a BOOL array is returned as some number of UDINT values (actually data type 0xD3).   If the BOOL is in a UDT, then it is different again.  I should really write up a wiki page just for BOOL.

timyhac, 2025-01-29:
In terms of my stance towards `Tag<M,T>`, there is a fairly lengthy discussion on [that issue](https://github.com/libplctag/libplctag.NET/issues/406). Looking at your listing above, you might find the code example on [this comment](https://github.com/libplctag/libplctag.NET/issues/406#issuecomment-2237996680)  relevant to you.

Were you able to use the advice above to work with the BOOL[0..15] tag?

zN3utr4l, 2025-01-29:
No, for now I'm on another project, as I said above I bypassed the problem by declaring the array with 32 positions, but monitoring only 16 of them, as soon as I have some time I think I'll stop using mappers and I'll try to find one generic solution to directly instantiate Tag().

Thanks for sharing the link to the discussion on Tag<M, T> I missed it.
Like the commenter on that thread, I also thought that Tag<M, T> was the best way to use the library, I tried to create a generic library for all PLCs by wrapping libplctag.NET.

timyhac, 2025-01-29:
Ok great - thanks!

closed 2025-01-29 by timyhac