libplctag/libplctag.NET #440: Tag.Abort is throwing an exception

VitorKawao, 2025-02-06:
Hello,

I am using libplctag at version 1.6.0-alpha.3 and libplctag.NativeImport at version 2.0.0.0-alpha.5

If I stop my thread that was reading, it throws an exception at Tag.Abort

CoreCLR Version: 6.0.3624.51421
.NET Version: 6.0.36
Description: The process was terminated due to an unhandled exception.
Exception Info: System.AggregateException: One or more errors occurred. (Cannot access a disposed object.
Object name: 'libplctag.Tag'.)
 ---> System.ObjectDisposedException: Cannot access a disposed object.
Object name: 'libplctag.Tag'.
   at libplctag.Tag.ThrowIfAlreadyDisposed() in /_/src/libplctag/Tag.cs:line 1085
   at libplctag.Tag.Abort() in /_/src/libplctag/Tag.cs:line 657
   at libplctag.Tag.<>c__DisplayClass134_0.<ReadAsync>b__0() in /_/src/libplctag/Tag.cs:line 817
   at System.Threading.CancellationToken.<>c.<Register>b__12_0(Object obj)
   at System.Threading.CancellationTokenSource.Invoke(Delegate d, Object state, CancellationTokenSource source)
   at System.Threading.CancellationTokenSource.CallbackNode.<>c.<ExecuteCallback>b__9_0(Object s)
   at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state)
--- End of stack trace from previous location ---
   at System.Threading.CancellationTokenSource.CallbackNode.ExecuteCallback()
   at System.Threading.CancellationTokenSource.ExecuteCallbackHandlers(Boolean throwOnFirstException)
   --- End of inner exception stack trace ---
   at System.Threading.CancellationTokenSource.ExecuteCallbackHandlers(Boolean throwOnFirstException)
   at System.Threading.CancellationTokenSource.NotifyCancellation(Boolean throwOnFirstException)
   at System.Threading.CancellationTokenSource.TimerCallback(Object state)
   at System.Threading.TimerQueueTimer.Fire(Boolean isThreadPool)
   at System.Threading.TimerQueue.FireNextTimers()

My logs
[log.txt](https://github.com/user-attachments/files/18694564/log.txt)

Thank you,
Vítor Guedes

timyhac, 2025-02-06:
Are you able to provide a repro?

VitorKawao, 2025-02-07:
My app has many modules executing simultaneously, each with its own thread.
One of these modules is the one that uses libplctag.
I can stop any module at any time.
If I understood correctly, when I stop it while I'm in the middle of a read, the app crashes at Tag.Abort()

CoreCLR Version: 6.0.3624.51421
.NET Version: 6.0.36
Description: The process was terminated due to an unhandled exception.
Exception Info: System.AggregateException: One or more errors occurred. (Cannot access a disposed object.
Object name: 'libplctag.Tag'.)
---> System.ObjectDisposedException: Cannot access a disposed object.
Object name: 'libplctag.Tag'.
at libplctag.Tag.ThrowIfAlreadyDisposed() in //src/libplctag/Tag.cs:line 1085
at libplctag.Tag.Abort() in //src/libplctag/Tag.cs:line 657
at libplctag.Tag.<>c__DisplayClass134_0.b__0() in /_/src/libplctag/Tag.cs:line 817
at System.Threading.CancellationToken.<>c.b__12_0(Object obj)
at System.Threading.CancellationTokenSource.Invoke(Delegate d, Object state, CancellationTokenSource source)
at System.Threading.CancellationTokenSource.CallbackNode.<>c.b__9_0(Object s)
at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state)
--- End of stack trace from previous location ---
at System.Threading.CancellationTokenSource.CallbackNode.ExecuteCallback()
at System.Threading.CancellationTokenSource.ExecuteCallbackHandlers(Boolean throwOnFirstException)
--- End of inner exception stack trace ---
at System.Threading.CancellationTokenSource.ExecuteCallbackHandlers(Boolean throwOnFirstException)
at System.Threading.CancellationTokenSource.NotifyCancellation(Boolean throwOnFirstException)
at System.Threading.CancellationTokenSource.TimerCallback(Object state)
at System.Threading.TimerQueueTimer.Fire(Boolean isThreadPool)
at System.Threading.TimerQueue.FireNextTimers()

Thank you,
Vítor Guedes

timyhac, 2025-02-07:
Just a simple repro would be very useful here.
I'm really not sure how you've developed your application.
Something like this would be useful enough for me to go on.

Note that the code below should not be copied or used as a reference implementation - for many reasons.
One reason is that it uses raw threads which should not be in normal C# application code.
I'm only using raw threads because you mentioned Thread in your comment.

Example format for Repro

```csharp  
var cts = new CancellationTokenSource();

Thread thread = new(() => ReadTagForever(cts.Token));
thread.Start();

Console.WriteLine("Press any key to end");
Console.ReadKey();
cts.Cancel();

void ReadTagForever(CancellationToken token)
{
    while (!token.IsCancellationRequested)
    {
        Tag tag = new()
        {
            Gateway = "127.0.0.1",
            Path = "1,0",
            Name = "MY_DINT[0]",
            PlcType = PlcType.ControlLogix,
            Protocol = Protocol.ab_eip,
        };

        tag.Read();
    }
}
```

timyhac, 2025-03-16:
Hi @VitorKawao - have you had a chance to create a repro of this issue?

VitorKawao, 2025-03-17:
No, I did not. But I fixed my problem by waiting some time to stop my thread. The problem occurred when I started a read (probably a write too) and aborted my thread before the read finished. Waiting resolved my issue

Thank you,
Vítor Guedes

aalex675, 2025-06-18:
I encountered this same problem and it is because of multithreaded use of the Read/Write/Abort/Dispose. The main library documentation does indicate that a mutex or similar locking mechanism should be used to control access to the tag object from multiple threads (https://github.com/libplctag/libplctag/wiki/API#controlling-multi-threaded-access-to-a-tag), but, for me at least, it wasn't obvious that the Dispose/Abort should also use the same locking mechanism.