libplctag/libplctag.NET #443: Make the size of the underlying tag hashtable in the C lib visible

iancolledge, 2025-05-06:
There's a function in src/utils/hashtable.c called hashtable_entries
It would be good if you make it visible from .NET
This is to make sure there's no problems with tags hanging around (bugs)

iancolledge, 2025-05-09:
Yep we'll do that no probs I'll close

closed 2025-05-09 by iancolledge

kyle-github, 2025-05-09:
Note that if you call `plc_tag_shutdown()`, the code will enumerate across the underlying array of the hash table and destroy all tags.  One of the tests I run is a hard shutdown where tags are still active and in use when shutdown is called.

kyle-github, 2025-05-09:
@iancolledge if you find any case that causes a memory leak, _please_ file an issue on the core library project!  I want to make sure those are fixes as soon as possible.