libplctag/libplctag.NET #449: Multiple instances running on same machine

BenGood, 2025-08-14:
I have a libplctag app that shows a few datapoints from a PLC. Sometimes I want to run a second instance of that app pointed at a different PLC but the second one doesn't seem to be able to connect. I'm not sure why, tag initialize doesn't throw an exception but afterward tag.initialized is false. 

Is this a limitation you know about or something I can work around?

timyhac, 2025-08-20:
Hi - I'm not aware of any reason this wouldn't work - are you able to provide [some debug logs and a code listing](https://github.com/libplctag/libplctag.NET/blob/main/CONTRIBUTING.md#guidance-when-creating-an-issue)?

timyhac, 2026-11-12:
Hi @BenGood - please raise a new ticket if you're able to expand on the problem.

closed 2026-11-12 by timyhac